microsoft/maker.js #384: Bug in result of makerjs.model.outline?

vespakoen, 2018-12-11:
Hi!

Great project!! This is going to be a match made in heaven for my openjscam project to generate g-code (https://github.com/openjscam/openjscam)

Now the problem I run into, when I create a model outline, and try to create another outline from the previous one.
Here is the code and outputs:

```js
import makerjs from 'makerjs'
const ellipse = new makerjs.models.Ellipse(100, 100)
console.log(JSON.stringify(ellipse, undefined, 2))
```
<details> 
    <summary>console.log(JSON.stringify(ellipse, undefined, 2))</summary>
    <p>

```json
{
  "models": {
    "Curve_1": {
      "type": "BezierCurve",
      "seed": {
        "type": "bezier-seed",
        "origin": [
          100,
          0
        ],
        "controls": [
          [
            100,
            26.52165
          ],
          [
            89.46432,
            51.95704
          ]
        ],
        "end": [
          70.71068,
          70.71068
        ]
      },
      "accuracy": 0.7853998908085255,
      "paths": {
        "arc_0": {
          "type": "arc",
          "origin": [
            0.000004751247953021368,
            0.0000057057216082842
          ],
          "radius": 99.99999524875221,
          "startAngle": 359.9999967308622,
          "endAngle": 44.99999961330228,
          "bezierData": {
            "startT": 0,
            "endT": 1
          }
        }
      }
    },
    "Curve_2": {
      "type": "BezierCurve",
      "seed": {
        "type": "bezier-seed",
        "origin": [
          70.71068,
          70.71068
        ],
        "controls": [
          [
[... 407 lines cut ...]
                    for (var pathId in modelContext.paths) {
---------------->   var pathContext = modelContext.paths[pathId];
                        if (!pathContext)
                            continue;
                        var walkedPath = {
                            modelContext: modelContext,
                            layer: (pathContext.layer != undefined) ? pathContext.layer : layer,
                            offset: newOffset,
                            pathContext: pathContext,
                            pathId: pathId,
                            route: route.concat(['paths', pathId]),
                            routeKey: routeKey + (routeKey ? '.' : '') + 'paths' + JSON.stringify([pathId])
                        };
                        if (options.onPath)
                            options.onPath(walkedPath);
                    }
                }
```

So it looks like the `insettedEllipse` does not have a `type` property anymore, it's just a bunch of arc's.
Is this valid? and if so, how would I go about insetting these again?

Thanks in advance!

- Koen

vespakoen, 2018-12-11:
Just debugged it a bit and went "back up the stack".
The issue seems to be coming from the joints, when using jointType 0, things seem to work normally.

vespakoen, 2018-12-11:
I just found out that it has to do with the endcaps, when I use join type `0` it works as expected, still something to look at and fix (throw a better error?)

danmarshall, 2018-12-11:
Hi @vespakoen - cool project! It does seem like a perfect fit. 
Thanks for submitting this bug. It does appear to be a bug that I need to track down. I'm glad you have a workaround for the meanwhile. 👍

closed 2018-12-11 by danmarshall

vespakoen, 2018-12-12:
Awesome thansks for the fix!

PR #385 (2018-12-11): fix for bevel
fixes #384