microsoft/maker.js #387: Having a hard time expanding complex shapes

vespakoen, 2018-12-12:
Making an outline on a complex shape (from a svg I imported into maker.js models) is very slow.
I tried simplifying the model, and converting to some keypoints and ConnectingTheDots, but it doesn't improve much, sometimes (probably due to the model) I get the following error:

```js
index.js:6439 Uncaught TypeError: Cannot read property 'valueIds' of undefined
```

[image]

vespakoen, 2018-12-12:
The model, (an imported Taj Mahal svg):

```json
{"models":{"path_0":{"paths":{"p_1":{"type":"line","origin":[19.472,-14.653],"end":[19.325000000000003,-9.958]},"p_2":{"type":"line","origin":[19.325000000000003,-9.958],"end":[19.25,-8.306]},"p_9":{"type":"line","origin":[18.549,-8.306],"end":[18.326,-14.655]},"p_10":{"type":"line","origin":[18.326,-14.655],"end":[17.141000000000002,-14.655]},"p_11":{"type":"line","origin":[17.141000000000002,-14.655],"end":[16.990000000000002,-9.905]},"p_18":{"type":"line","origin":[16.422,-9.905],"end":[16.275000000000002,-14.652999999999999]},"p_19":{"type":"line","origin":[16.275000000000002,-14.652999999999999],"end":[15.247000000000002,-14.652999999999999]},"p_20":{"type":"line","origin":[15.247000000000002,-14.652999999999999],"end":[15.247000000000002,-11.252999999999998]},"p_21":{"type":"line","origin":[15.247000000000002,-11.252999999999998],"end":[15.157000000000002,-11.242999999999999]},"p_22":{"type":"line","origin":[15.157000000000002,-11.242999999999999],"end":[15.157000000000002,-10.596999999999998]},"p_23":{"type":"line","origin":[15.157000000000002,-10.596999999999998],"end":[14.519000000000002,-10.596999999999998]},"p_24":{"type":"line","origin":[14.519000000000002,-10.596999999999998],"end":[14.519000000000002,-11.172999999999998]},"p_25":{"type":"line","origin":[14.519000000000002,-11.172999999999998],"end":[13.913000000000002,-11.168999999999999]},"p_26":{"type":"line","origin":[13.913000000000002,-11.168999999999999],"end":[13.843000000000002,-11.168999999999999]},"p_27":{"type":"line","origin":[13.843000000000002,-11.168999999999999],"end":[13.843000000000002,-10.181]},"p_28":{"type":"line","origin":[13.843000000000002,-10.181],"end":[13.634000000000002,-10.181]},"p_35":{"type":"line","origin":[12.168000000000005,-9.857999999999999],"end":[12.168000000000005,-9.691]},"p_36":{"type":"line","origin":[12.168000000000005,-9.691],"end":[11.893000000000004,-9.691]},"p_40":{"type":"line","origin":[10.061000000000005,-4.967000000000001],"end":[10.041000000000006,-4.967000000000001]},"p_43":{"type":"line","origin":[9.945000000000006,-3.7790000000000012],"end":[9.945000000000006,-3.774]},"p_45":{"type":"line","origin":[9.921,-3.746],"end":[9.919,-3.746]},"p_46":{"type":"line","origin":[9.919,-3.746],"end":[9.917,-3.746]},"p_48":{"type":"line","origin":[9.895999999999999,-3.774],"end":[9.895999999999999,-3.779]},"p_51":{"type":"line","origin":[9.799999999999999,-4.967],"end":[9.778,-4.967]},"p_55":{"type":"line","origin":[7.947,-9.69],"end":[7.671,-9.69]},"p_56":{"type":"line","origin":[7.671,-9.69],"end":[7.671,-9.857999999999999]},"p_63":{"type":"line","origin":[6.207,-10.183],"end":[5.997,-10.183]},"p_64":{"type":"line","origin":[5.997,-10.183],"end":[5.997,-11.161]},"p_65":{"type":"line","origin":[5.997,-11.161],"end":[5.321,-11.174]},"p_66":{"type":"line","origin":[5.321,-11.174],"end":[5.321,-10.597999999999999]},"p_67":{"type":"line","origin":[5.321,-10.597999999999999],"end":[4.681,-10.597999999999999]},"p_68":{"type":"line","origin":[4.681,-10.597999999999999],"end":[4.681,-11.244]},"p_69":{"type":"line","origin":[4.681,-11.244],"end":[4.594,-11.254]},"p_70":{"type":"line","origin":[4.594,-11.254],"end":[4.594,-14.654]},"p_71":{"type":"line","origin":[4.594,-14.654],"end":[3.563,-14.654]},"p_72":{"type":"line","origin":[3.563,-14.654],"end":[3.417,-9.906]},"p_79":{"type":"line","origin":[2.846,-9.906],"end":[2.697,-14.654]},"p_80":{"type":"line","origin":[2.697,-14.654],"end":[1.513,-14.654]},"p_81":{"type":"line","origin":[1.513,-14.654],"end":[1.29,-8.306]},"p_88":{"type":"line","origin":[0.59,-8.306],"end":[0.515,-9.96]},"p_89":{"type":"line","origin":[0.515,-9.96],"end":[0.368,-14.655000000000001]},"p_90":{"type":"line","origin":[0.368,-14.655000000000001],"end":[0,-14.655000000000001]},"p_91":{"type":"line","origin":[0,-14.655000000000001],"end":[0,-16.094]},"p_92":{"type":"line","origin":[0,-16.094],"end":[19.84,-16.094]},"p_93":{"type":"line","origin":[19.84,-16.094],"end":[19.84,-14.655000000000001]},"p_94":{"type":"line","origin":[19.84,-14.655000000000001],"end":[19.472,-14.655000000000001]},"p_95":{"type":"line","origin":[19.472,-14.655000000000001],"end":[19.472,-14.653]},"p_96":{"type":"line","origin":[10.72,-14.636],"end":[9.119,-14.636]},"p_97":{"type":"line","origin":[9.119,-14.636],"end":[9.119,-12.940999999999999]},"p_102":{"type":"line","origin":[10.72,-12.940999999999999],"end":[10.72,-14.636]}},"models":{"p_3":{"type":"BezierCurve","accuracy":0.0025703699150567165,"seed":{"type":"bezier-seed","origin":[19.25,-8.306],"controls":[[19.26,-8.254999999999999],[19.274,-8.152999999999999]],"end":[19.229,-8.053999999999998]},"paths":{"arc_0":{"type":"arc","origin":[18.611133320251096,-8.18067194892167],"radius":0.6510435890787049,"startAngle":348.90108884768796,"endAngle":349.58430581942804,"bezierData":{"startT":0,"endT":0.047619075}},"arc_1":{"type":"arc","origin":[18.642139271032818,-8.186418373248154],"radius":0.6195096441668904,"startAngle":349.5885799234849,"endAngle":349.8414665021802,"bezierData":{"startT":0.047619075,"endT":0.0634921}},"arc_2":{"type":"arc","origin":[18.7019177664804,-8.196860430821117],"radius":0.5588264687242608,"startAngle":349.8143051752302,"endAngle":352.6922498076533,"bezierData":{"startT":0.0634921,"endT":0.2084772359375}},"arc_3":{"type":"arc","origin":[18.7597332121568,-8.204212822330389],"radius":0.5005454344717818,"startAngle":352.68522336554264,"endAngle":356.9119793184935,"bezierData":{"start [... cut ...] 4404,"bezierData":{"startT":0.9554455,"endT":0.9749381375}},"arc_9":{"type":"arc","origin":[10.553974877420739,-12.938738672068801],"radius":0.16604174382260814,"startAngle":359.4509679251318,"endAngle":359.5120173591358,"bezierData":{"startT":0.9749381375,"endT":0.9777228}},"arc_10":{"type":"arc","origin":[10.633493062466407,-12.939514124107403],"radius":0.08651971482041969,"startAngle":359.0439311111117,"endAngle":359.45987598965877,"bezierData":{"startT":0.9777228,"endT":0.9944307}}}}}}}}
```

danmarshall, 2018-12-12:
Hi @vespakoen , can you provide the value which causes the error?
I do have a work item to create a better algorithm for this. It's been on my list for a while...

forresto, 2019-10-22:
Here's a path that fails with `distance = 10`, but works with `5`:

```
"M221.59 49.6C245.93 49.6 259.93 65 259.93 79.5C259.93 81.8869 259.8 84.4147 259.305 87H277.002L282.34 42.8C285.43 17.2 301.46 0 321.18 0L312.62 69.3V69.4C322.18 57.5 333.53 49.4 348.37 49.4C364.41 49.4 380.44 59.9 380.64 93.1C380.64 99.7368 379.882 106.453 378.44 113H400.629L405.14 75.6H389.6L390.1 71.6C391.5 60.7 399.86 52.1 407.43 52.1H407.93L408.03 51.4C411.78 20.6 426.42 0.4 473.03 0.7C473.03 0.7 470.84 13.2 470.14 16.8C469.25 21.4 465.66 25.6 457.59 26.4C445.77 27.7 442.19 35.7 440.59 50.9L440.49 52.1H463.7L463 56.4C462.891 57.2936 462.744 58.1606 462.562 59H478.412L485.61 0.7C500.65 0.7 517.08 17.9 513.67 45.6L505.31 113H522.555C522.896 108.966 523.566 104.937 524.55 101C532.32 69.9 557.61 50.1 584.4 50.1C606.01 50.1 625.372 63.5645 629.261 89H645.403L639.38 52.1C655 52.1 674.34 65 675.78 86.2C675.78 86.2 678.17 128.5 678.27 131.8C679.66 128.2 696 85.7 696 85.7C702.42 68.4 714.67 52 737.58 52C737.58 52 747.44 128.1 747.83 131.7C748.93 128.2 762.37 85.8 762.37 85.8C768.75 68.4 781.3 52.1 805 52.1L771.22 135.4C764.22 152.1 751.32 167.4 727.22 167.4C727.22 167.4 716.96 96.1 716.86 94.5C716.36 96.2 701.92 133.7 701.92 133.7C695.15 151 682.2 167.3 658.2 167.4L646.382 95H629.874C629.939 96.21 629.971 97.4434 629.97 98.7C629.97 130.6 607.39 168.9 567.97 168.9C536.432 168.9 522.593 145.346 522.298 120H504.442L502.42 136.3C499.63 159.6 484.68 178.1 463.68 178.4C463.18 178.4 462.58 178.3 462.09 178.3L466.09 158.8L477.795 64H460.946C457.691 71.3735 451.493 75.6 446 75.6H438.7L428.52 160.1C425.44 185.2 411.49 202.5 389.68 203.3C389.18 203.3 388.59 203.2 388.09 203.2L392.09 183.8L399.785 120H376.619C368.682 146.361 349.58 168.3 324.57 168.3C311.72 168.3 303.06 162.3 297.28 154.8C290.81 163.2 278.76 166.5 267.4 166.5L276.156 94H256.937C251.346 104.964 235.667 115.631 192.61 119.9C192.61 138.2 202.57 142.6 212.33 142.6C217.791 142.772 223.197 141.459 227.97 138.8C230.85 137.3 245.89 130.5 257.54 145.2C244 158.4 230.76 168.3 205.46 168.3C182.393 168.3 159.492 153.337 159.184 120H138.241L132 135.4C125.05 152.1 112.1 167.4 88 167.4C88 167.4 77.78 96.1 77.64 94.5C77.08 96.2 62.64 133.7 62.64 133.7C55.87 151 42.92 167.3 18.92 167.4L0 52.1C15.64 52.1 35 65 36.35 86.2C36.35 86.2 38.74 128.5 38.84 131.8C40.24 128.2 56.57 85.7 56.57 85.7C63 68.4 75.29 52 98.2 52C98.2 52 108.06 128.1 108.46 131.7C109.55 128.2 123 85.8 123 85.8C129.47 68.4 142 52.1 165.76 52.1L141.078 113H159.437C159.521 111.949 159.625 110.883 159.75 109.8C164.33 69.7 194.3 49.6 221.59 49.6Z"
```

Live demo: https://observablehq.com/d/a39a12564cd5b6ac

danmarshall, 2019-10-23:
Thanks @forresto ! - I will use that as a test case in an upcoming change.

MartinSoderholm, 2020-02-27:
Hi, having the same issue with models from svg. Is this being worked on?

Is there perhaps a workaround to avoid the error by simplifying or limiting the svg path commands in some way?

Thanks

danmarshall, 2020-02-27:
Hi @MartinSoderholm , I've been working on a solution for quite some time now. If you are feeling daring you can try a build of the [combine-sweep branch](https://github.com/microsoft/maker.js/tree/combine-sweep) and see how it works in your case. Fair warning - this latest algorithm still has its issues. I do hope to have some time to finalize it soon.

MartinSoderholm, 2020-02-27:
Thanks @danmarshall, we'll have a look at it 👍

danmarshall, 2020-02-27:
@MartinSoderholm also, can you tell me the nature of your SVG path? e.g. is it made up of mostly Beziers (commands like C, Q, S, T) ?

MartinSoderholm, 2020-02-27:
Mostly curves.
```svg
<?xml version="1.0" encoding="utf-8"?>
<svg version="1.1" id="Layer_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink"
  viewBox="0 0 190.4 283.5" xml:space="preserve">
  <path d="M180.1,166.7c-11.4-20.3-26.7-27.8-31.9-30.7c15.5-8.3,27.6-22.1,33.7-38.5c5.3-14.3,6.1-29,2.2-45
    C178.3,29,162.3,16.1,153.3,10c-3.4-2.3-17-10-27.9-10H0v283.5h121c15.9,0,34-10.4,43-18.5c7.1-6.3,11.3-12.1,17.2-22
    s9.1-24.7,9.1-37C190.3,187.5,184.5,174.6,180.1,166.7z M104.9,37.6c9.2,0,26.5,3.9,34,22.9c3,7.5,3.7,17.9,0.6,26.9
    c-3.4,10.4-15.1,24.7-33.1,24.7H44.8V37.6H104.9z M138.4,214.2c-5.2,12.3-18.4,23.1-32.6,23.1H44.7v-77.9H106c8,0,17.1,4.7,22.5,9.7
    C140.7,180.3,145.1,198.4,138.4,214.2z"/>
</svg>
```