microsoft/maker.js #388: Export to PDF

icodk, 2018-12-15:
Could anybody help with  how to export a model to PDF ?
I am using vuejs and have difficulty in 'importing' PDFKit, which can be downloaded as a js library from https://github.com/foliojs/pdfkit/releases

Tried something like:
`npm install pdfkit`

then in index.html
```
<script>
    var pdfKit=  require ('pdfkit');
    </script>
```
Then in vue component
```
var doc = new pdfkit();
this.values.PDFExport=makerjs.exporter.toPDF(doc,model);
```
However already in index.html I got this error:
`browser.maker.js:71 Uncaught Error: Cannot find module 'pdfkit'`
Thanks in advance

vespakoen, 2018-12-15:
In your index.html the variable is called `pdfKit` (with capital K), but your vue component it is not using an upper case, also, I think you cannot use "require"  in a <script> block, you might have to put that in a seperate file an use some compiler (I assume you already have this though), you can try to put the require in the vue component file

```js
var pdfkit=  require('pdfkit');
var doc = new pdfkit();
this.values.PDFExport=makerjs.exporter.toPDF(doc,model);
```

icodk, 2018-12-16:
I am almost there (hopefully)
Here is what I did so far:
I used browserify to bundle pdfkit.js:
`browserify  -r pdfkit >pdfkit_bundle.js`
In index.html I added
`<script src='pdfkit_bundle.js'></script>`
and 
 ```
<script>
      var pdfkit=  require ('pdfkit');
  </script>
```
In my component I try to use:
```
var doc = new pdfkit();
makerjs.exporter.toPDF(doc,model);
this.values.PDFExport=doc;
```
However I get the following error from the `makerjs.exporter.toPDF(doc,model);`

`vue.esm.js?efeb:591 [Vue warn]: Error in mounted hook: "TypeError: Cannot read property 'origin' of undefined"
`
Both the model and the doc are  have a value so they are NOT undefined
My model is otherwise fine and I can export it to  DXF without an issue and show it on the page
Any idea ?

danmarshall, 2018-12-16:
@icodk can you go into your debugger, and enable `Pause on exceptions` and `Pause on caught exceptions` and see the exact line of code causing the error?

icodk, 2018-12-16:
@danmarshall here is the line that causes the exception, it is inside 
` function toPDF(doc, modelToExport, options)`
about line 7208 in browser.maker.js as far as I can see

        `offset = MakerJs.point.add(offset, options.origin);`
The error is probably about the options parameter that according to the doc is optional
I could try to supply such option object but I am not sure how to define such object

danmarshall, 2018-12-16:
@icodk - that's a bug that should be fixed. In the meanwhile, you can just send an empty object `{}` - all the properties are optional:
https://maker.js.org/docs/api/interfaces/makerjs.exporter.ipdfrenderoptions.html

icodk, 2018-12-17:
@danmarshall This works (almost) : Downloading the doc as I do with the DXF results  a file with the text:
`[object PDFDocument]`
That's all
here is my code:
```
 var doc = new pdfkit();
        var pdf_opt={};
        makerjs.exporter.toPDF(doc,model,pdf_opt);
        var element = document.createElement('a');
        element.setAttribute('href', 'data:application/pdf;charset=utf-8,' + encodeURIComponent(doc));
        element.setAttribute('download', 'mydoc.pdf');
         element.click();    
```
Not sure where the error, Is it in maker or the way I download it

danmarshall, 2018-12-17:
Hi @icodk - the DXF works a little differently, since it is plain text. See the [PDFKit documentation](http://pdfkit.org/docs/getting_started.html#using_pdfkit_in_the_browser) on how to create a blob url for download. You can also take a look at the [Playground code](https://github.com/Microsoft/maker.js/blob/master/packages/playground/src/worker/export-worker.ts#L94) and see another way to do it, which involves [implementing the Node.js string reader interface](https://github.com/Microsoft/maker.js/blob/master/packages/playground/src/worker/string-reader.ts).

icodk, 2018-12-18:
Hi @danmarshall  Thanks for your patient.
finally working
here is some code for others (may be to add something to the docs, was not clear to me how it works):

On commandline:
```
npm install  blob-stream --save
browserify   -r blob-stream > blob_stream.js
```
in index.html (in the <head>:
```
<script src='blob_stream.js'></script>

 <script>
    var makerjs=require('makerjs');
    var pdfkit=  require ('pdfkit');
    var blobStream=  require ('blob-stream');
 </script>

```
Generating the pdf
```   
var doc = new pdfkit();
        var stream = doc.pipe(blobStream());
        var pdf_opt={};
        makerjs.exporter.toPDF(doc,mymodel,pdf_opt);
        doc.end();

        stream.on('finish',function(){
            var element = document.createElement('a');
            var blobURL=stream.toBlobURL('aplication/pdf');
           element.setAttribute('href', blobURL);
          element.setAttribute('download', 'mydoc.pdf');
          element.click();
      });      

```

closed 2019-01-10 by danmarshall