microsoft/maker.js #389: [suggestion] Expand $() functions abilities

vespakoen, 2019-01-02:
Happy new year!

I was looking at cadquery (https://dcowden.github.io/cadquery/apireference.html#d-operations) and thought their API is quite nice, I'd like to suggest extending it's functionality to include things like `fillet`, `dogbone` and the drawing functions that remember the "current position", like `moveTo`, `lineTo`, `lineH`, `lineV` etc.

danmarshall, 2019-01-10:
Hey @vespakoen Happy new year to you! That is a pretty cool API. Would be cool to make it a standalone project outside of MakerCam so that it can be used in other projects.

danmarshall, 2019-01-17:
@vespakoen some of these commands such as `moveTo`, `lineTo` etc. might be good for a new `ChainBuilder` class. But that idea would only be a subset of what your example contains.