microsoft/maker.js #391: I’m having trouble combining some expanded paths

michaelrhodes, 2019-01-24:
I’m starting with an SVG path that defines the letter A as pictured below:
```js
var a = makerjs.importer.fromSVGPathData('M47,112Q47,112,51,107Q54,102,58,97Q62,93,67,85Q72,77,78,66Q83,56,84,54Q84,52,85,51Q86,50,86,49Q86,48,87,48Q87,48,87,56Q87,64,88,67L87,106 M65,90Q65,90,65,89Q65,89,68,88Q71,87,78,86L93,82')
```
[image]

I’m able to expand its paths and get the following:
```js
makerjs.model.expandPaths(a, 4)
```
[image]

However when I attempt to fill the path I see:
[image]

I know`fill-rule: evenodd` would visually do the trick, but my intention is to use the expanded path as a font glyph, so I need to figure out how to recreate Illustrator’s “exclude” operation to achieve:
[image]

I tried all the configurations of `makerjs.model.combine` with all the `a` and `b` arguments I could think to try, but to no avail. I can go back and record specific examples if need be, but I’m hoping there’s an obvious solution.

danmarshall, 2019-01-24:
Hi @michaelrhodes you're combing correctly. The `fill-rule` comes into play when you export. See https://maker.js.org/docs/exporting/#SVG for options, I think you will want to set `fillRule: "nonzero"` as an option.
Also, please let me know how we can make this easier to discover in the documentation :)

michaelrhodes, 2019-01-24:
Thanks _so much_ @danmarshall, that was exactly the issue!

I actually tried `exporter.toSVG` with `useSvgPathOnly: false` at some point, so I was nearby the `fillRule` documentation, but I assumed that property would just change the attribute value on the exported element. That said, in retrospect there were two things that suggested otherwise. One was the “requires more computation” note, but maybe that meant more computation for the renderer, not the library? The other was that the `fillRule` option also appeared in the API documentation for `exporter.toSVGPathData`. This is a long-winded way of saying I’m not really sure what would have made it easier, outside of a specific example of exporting objects with overlapping paths.

Thanks again!

closed 2019-01-24 by michaelrhodes