microsoft/maker.js #393: layer name & layer colour in DXF export

timdiacon, 2019-02-19:
It would be great to be able to specify an object for the layers in the DXF object which detaches the name from the colour... something like

`layerOptions: {
color:'red',
name:'10mm outline'
}`

danmarshall, 2019-02-23:
Hi @timdiacon , I believe this is already possible with [layerOptions](https://maker.js.org/docs/api/interfaces/makerjs.exporter.idxfrenderoptions.html#layeroptions) which is an object, where each layer name can be a key and the value is an object with options for that layer. Currently the only option to set is `color`. 
 
So in your case, you would call:
```
makerjs.exporter.toDXF(yourModel, {
  layerOptions: {
    "10mm outline": {
      color: "red"
    } 
  }
});
```
Looks like this should be added to the documentation. Pull requests are welcome!

timdiacon, 2019-03-18:
Hmm I've tried this but not getting any joy. Looking at the api documentation is seems that the DXF layer color is expecting a number?

`export interface IDXFLayerOptions {

        /**
         * DXF layer color.
         */
        color: number
    }`

Rather than a string? I've tried passing 0xff00ff but that also didn't work work. Is it expecting a number which correlates to an array position or something perhaps?

danmarshall, 2019-03-18:
Oh that’s right, sorry. To get the number, try using `makerjs.exporter.color.red` instead of the string ‘red’.
See the list of colors here:
https://github.com/Microsoft/maker.js/blob/b393c682/packages/maker.js/src/core/exporter.ts#L68

timdiacon, 2019-03-18:
Booom! Yep that's working although it's colors not color :-)

`this.dxf = makerjs.exporter.toDXF(this.model, {
					layerOptions: {
						"10mm outline": {
							color: makerjs.exporter.colors.red
						} 
					}
				});`

closed 2019-03-18 by timdiacon

PodgPow, 2019-05-07:
I changed the library to export to DXF. Now setting the color and name of the layer is done very simply.
Example:
Dots = new makerjs.models.ConnectTheDots("Array with coordinates");
Dots.layer = ["Name layer", "color layer"];

PodgPow, 2019-05-13:
I changed the file index.js in my project for it.  And how to build a project-I don't know - errors (issue #405)