microsoft/maker.js #394: Option to not convert on fromSVGPathData import

jakecausier, 2019-03-18:
Hi Dan, fantastic piece of work MakerJS is, we've been using it for a while to help build some CAD tools.

One question I had was if it's possible to prevent the `fromSVGPathData` importer from converting the original SVG syntax into separate lines/arcs etc.? We have tried `useSvgPathOnly` option in the SVG exporting however it breaks everything else in our current rendering. We'd just like to keep the path the same as-is throughout the render, as one single object.

A test case:
> M0,100 L0,100 L0, -1050 L4300, -1050 L4300, 100 L4150, 100 L4150, -900 L150, -900 L150, 100 Z

With `useSvgPathOnly` to `true`, this will retain the original path data but break everything else for us. Set to `false`, and it will break it down into simple line elements and do this globally to everything.

If possible, we'd like to propose fromSVGPathData have an extra option in the ISVGImportOptions interface for either converting to individual elements (default) or to retain the structure as-is for this single element.

danmarshall, 2019-03-18:
Hello @jakecausier thank you for the kind words. If you would like to get the exact output the same as your input, you may want to just omit that particular model during your export, then modify the SVG after your export. Here is an example:

```
var d = 'M0,100 L0,100 L0, -1050 L4300, -1050 L4300, 100 L4150, 100 L4150, -900 L150, -900 L150, 100 Z';
var drawing = {
    models: {
        ring: new makerjs.models.Ring(500, 300),
        square: makerjs.model.center(new makerjs.models.Square(1300)),
        ignore: makerjs.importer.fromSVGPathData(d)
    }
};
var drawingIgnored = {
    models: Object.assign({}, drawing.models, { ignore: null })
};
var svg = makerjs.exporter.toSVG(drawing);
var svgIgnored = makerjs.exporter.toSVG(drawingIgnored);
//now replace the path data from svg with the path data from svgIgnored
console.log(svg);
console.log(svgIgnored);
var ignoredPathData = /path d="(.*?)"/.exec(svgIgnored)[1];
var newSvg = svg.replace(/path d="(.*?)"/, `path d="${d} ${ignoredPathData}"`);
console.log(newSvg);
```

jakecausier, 2019-03-20:
Thanks for the solution @danmarshall, it works ideally.

We had another look at the code we wrote and realised `useSvgPathOnly` actually works pretty well... our scale was just off and after some caption resizing it works like a dream.

closed 2019-03-20 by jakecausier