microsoft/maker.js #398: How to create in makerjs create a grid and scale in the documentation?

doox911, 2019-04-23:
I can not find the documentation on makerjs, how to implement the scale and grid in the examples of documentation?

danmarshall, 2019-04-23:
Hi @doox911, are you referring to the blue UI as in the demo page? That is the Playground app. Others have been wanting this also, please see https://github.com/Microsoft/maker.js/issues/380#issuecomment-447597337

doox911, 2019-04-23:
Thanks for the answer, but I did not find the definition of the grid and scale functions.

danmarshall, 2019-04-23:
I still want to make sure we are talking about the blue canvas in the Playground app. I'm not sure if you might mean [grid layout](https://maker.js.org/docs/intermediate-drawing/#Repeating%20layouts) and [scale](https://maker.js.org/docs/intermediate-drawing/#Scaling).

doox911, 2019-04-23:
I am interested in this:
[image]

danmarshall, 2019-04-25:
Unfortunately, the [Playground code](https://github.com/Microsoft/maker.js/blob/master/packages/playground/src/playground.ts) is currently monolithic and is in need of componentization to allow using just parts of it. However, there isn't much magic to scaling to fit. Maybe look at [this function](https://github.com/Microsoft/maker.js/blob/master/packages/playground/src/playground.ts#L1362)

lautr3k, 2019-11-10:
Hello, I also need the beautiful blueprint for an ongoing project. So I made this little module. If it helps :) https://onlfait.github.io/svg-blueprint/docs/

doox911, 2019-11-27:
> Hello, I also need the beautiful blueprint for an ongoing project. So I made this little module. If it helps :) https://onlfait.github.io/svg-blueprint/docs/

Thanks. I will definitely try it. I wrote you a issues.