microsoft/maker.js #399: How to draw an arch with offset center of a circle?

doox911, 2019-04-23:
No problem to draw such an arch:
[image]

But how to draw the offset circle in `maker`?

[image]

danmarshall, 2019-04-23:
The easiest way may be to [create an arc from 3 points](https://maker.js.org/docs/api/classes/makerjs.paths.arc.html#constructor).

danmarshall, 2019-04-23:
```
        /**
         * Class for arc path, created from 3 points.
         * 
         * @param pointA First end point of the arc.
         * @param pointB Middle point on the arc.
         * @param pointC Second end point of the arc.
         */
        constructor(pointA: IPoint, pointB: IPoint, pointC: IPoint);
```

doox911, 2019-04-24:
I'm sorry, but I can't understand what is the center, what x and y (

doox911, 2019-04-24:
Спасибо)

closed 2019-04-24 by doox911