microsoft/maker.js #400: how to draw double line in makerjs?

Mrluobo, 2019-04-24:
Hi ,i am new to makerjs. we often use double line in autoCAD to means walls.I want to draw double line using makerjs and export cxf file. is there any easy way to draw double line in makerjs?

hope for you help   thx

danmarshall, 2019-04-24:
Yes, you can use the [Wireframe technique](https://maker.js.org/docs/advanced-drawing/#Wireframe%20technique). First, make a single line drawing. Then, call [`makerjs.model.expandPaths`](https://maker.js.org/docs/api/modules/makerjs.model.html#expandpaths).
See [Expanding paths](https://maker.js.org/docs/advanced-drawing/#Expanding%20paths) for more info.

Mrluobo, 2019-04-25:
it works well!   thank you so much

closed 2019-04-25 by Mrluobo