microsoft/maker.js #402: How in Maker to save in a variable only the generated path (D)?

doox911, 2019-04-26:
How in Maker to save in a variable only the generated path (D)?

danmarshall, 2019-04-26:
Try [`makerjs.exporter.toSVGPathData(model)`](https://maker.js.org/docs/api/modules/makerjs.exporter.html#tosvgpathdata) - sorry this is not in the tutorial, but it is in the API documentation.

doox911, 2019-04-26:
Thanks. I just found it myself) I'm sorry.

danmarshall, 2019-04-26:
No worries! I'd like to write the page for the tutorial someday. Or I can take a pull request :)

doox911, 2019-04-26:
example:

```
this.model = {
          models : {
            dome : new makerjs.models.Dome(50, 30),
            arch : {
              paths: {
                "line1": new makerjs.paths.Line([0, 0], [0, +this.height]),
                "arc": new makerjs.paths.Arc([0, +this.height], [+this.horizontal, +this.height + (+this.horizontal)], [+this.horizontal*2, +this.height]),
                "line2": new makerjs.paths.Line([+this.horizontal*2, +this.height], [+this.horizontal*2, 0]),
                "line3": new makerjs.paths.Line([+this.horizontal*2, 0], [0, 0])
              },
            }
          }
        }; 
```
How get two path?

danmarshall, 2019-04-26:
call it twice:
```
var d_dome = makerjs.exporter.toSVGPathData(this.model.models.dome);
var d_arch = makerjs.exporter.toSVGPathData(this.model.models.arch);
```

doox911, 2019-04-26:
Very Good!) Thanks!

closed 2019-04-26 by doox911