microsoft/maker.js #404: DXF export

pete273, 2019-05-07:
We have developed a 2D web based drawing app using the paper.js library. This uses maker.js to export  a drawing to dxf format for use with a laser cutter.
The problem is that the laser cutter software used in UK schools will not open the file.  However there is an old program (Techsoft 2D designer) that will read the file and this then exports a dxf that works with the laser cutter software.
Do you have any suggestions?

danmarshall, 2019-05-07:
Hi @pete273 - I have also seen that some programs are able to open the Maker.js DXF and others are not. I would also like to know why. How about we compare the DXF between the same drawing? Below is the DXF for a [simple square](https://maker.js.org/playground/?script=Square). Can you click this link, generate the DXF, import it into your program, then export DXF again, and paste the text in a new comment? Then we can compare the two.

```
0
SECTION
2
HEADER
9
$INSUNITS
70
4
0
ENDSEC
0
SECTION
2
TABLES
0
TABLE
2
LAYER
0
ENDTAB
0
ENDSEC
0
SECTION
2
ENTITIES
0
LINE
8
0
10
0
20
0
11
100
21
0
0
LINE
8
0
10
100
20
0
11
100
21
100
0
LINE
8
0
10
100
20
100
11
0
21
100
0
LINE
8
0
10
0
20
100
11
0
21
0
0
ENDSEC
0
EOF
```

focusedsoft, 2019-05-08:
Dan, many thanks for replying to Pete. We've asked a school teacher to import the square.dxf file generated in Maker.js into the Techsoft 2D Designer program then export it again. Pete or I will paste the new file here as soon as it comes through.

pete273, 2019-05-08:
Here's the export of the square dxf form the TechSoft program. As you can see it's considerably different.

0
SECTION
2
HEADER
9
$ANGDIR
70
0
9
$CHAMFERA
40
10
9
$CHAMFERB
40
10
9
$CLAYER
8
Layer_1
9
$EXTMAX
10
260
20
199
9
$EXTMIN
10
160
20
99
9
$LIMMAX
10
420
20
297
9
$LIMMIN
10
0
20
0
9
$LTSCALE
40
10
9
$PLINEGEN
70
0
9
$TEXTSIZE
40
20
9
$GRIDUNIT
10
10
20
10
9
$SNAPUNIT
10
1
20
1
0
ENDSEC
0
SECTION
2
TABLES
0
TABLE
2
STYLE
70
1
0
STYLE
2
STANDARD
70
0
40
0
41
1
50
0
71
0
42
0.20000000000000001110223
3
txt
4

0
ENDTAB
0
TABLE
2
LTYPE
70
14
0
LTYPE
2
CONTINUOUS
70
0
3
Solid line
72
65
73
0
40
0
0
LTYPE
2
DOT
70
0
3
............................................
72
65
73
2
40
0.25
49
0
49
-0.25
0
LTYPE
2
HIDDEN
70
0
3
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
72
65
73
2
40
0.375
49
0.25
49
-0.125
0
LTYPE
2
TSLONGDASH
70
0
3
__ __ __ __ __ __ __ __ __ __ __ __ __ __ _ 
72
65
73
2
40
1.28000000000000002664535
49
1
49
-0.280000000000000026645353
0
LTYPE
2
TSDASHDOT
70
0
3
______ . ______ . ______ . ______ . ______ .
72
65
73
4
40
1.28000000000000002664535
49
1
49
-0.140000000000000013322676
49
0
49
-0.140000000000000013322676
0
LTYPE
2
TSCENTER
70
0
3
______ _ ______ _ ______ _ ______ _ ______ _
72
65
73
4
40
2
49
1.5
49
-0.179999999999999993338662
49
0.140000000000000013322676
49
-0.179999999999999993338662
0
LTYPE
2
TSPHANTOM
70
0
3
_____ _ _ _____ _ _ _____ _ _ _____ _ _ ____
72
65
73
6
40
2.5
49
1.62999999999999989341859
49
-0.190000000000000002220446
49
0.149999999999999994448885
49
-0.190000000000000002220446
49
0.149999999999999994448885
49
-0.190000000000000002220446
0
LTYPE
2
DASHED
70
0
3
__ __ __ __ __ __ __ __ __ __ __ __ __ __ __
72
65
73
2
40
0.75
49
0.5
49
-0.25
0
LTYPE
2
CENTER
70
0
3
____ _ ____ _ ____ _ ____ _ ____ _ ____ _ __
72
65
73
4
40
2
49
1.25
49
-0.25
49
0.25
49
-0.25
0
LTYPE
2
PHANTOM
70
0
3
_____ _ _ _____ _ _ _____ _ _ _____ _ _ ____
72
65
73
6
40
2.5
49
1.25
49
-0.25
49
0.25
49
-0.25
49
0.25
49
-0.25
0
LTYPE
2
DASHDOT
70
0
3
__ . __ . __ . __ . __ . __ . __ . __ . __ .
72
65
73
4
40
1
49
0.5
49
-0.25
49
0
49
-0.25
0
LTYPE
2
BORDER
70
0
3
__ __ . __ __ . __ __ . __ __ . __ __ . __ _
72
65
73
6
40
1.75
49
0.5
49
-0.25
49
0.5
49
-0.25
49
0
49
-0.25
0
LTYPE
2
DIVIDE
70
0
3
__ . . __ . . __ . . __ . . __ . . __ . . __
72
65
73
6
40
1.25
49
0.5
49
-0.25
49
0
49
-0.25
49
0
49
-0.25
0
ENDTAB
0
TABLE
2
LAYER
70
3
0
LAYER
2
Layer_1
70
0
62
7
6
CONTINUOUS
0
LAYER
2
Layer_2
70
0
62
7
6
CONTINUOUS
0
LAYER
2
Layer_3
70
0
62
7
6
CONTINUOUS
0
ENDTAB
0
ENDSEC
0
SECTION
2
ENTITIES
0
LINE
8
Layer_1
6
CONTINUOUS
62
7
10
160
20
99
11
260
21
99
0
LINE
8
Layer_1
6
CONTINUOUS
62
7
10
260
20
99
11
260
21
199
0
LINE
8
Layer_1
6
CONTINUOUS
62
7
10
260
20
199
11
160
21
199
0
LINE
8
Layer_1
6
CONTINUOUS
62
7
10
160
20
199
11
160
21
99
0
ENDSEC
0
EOF

danmarshall, 2019-05-08:
Hi @pete273 & @focusedsoft - you mentioned 
> laser cutter software used in UK schools

Can you tell me what this is, maybe I can run some tests to try it out. Otherwise this process might take a long time ;)

focusedsoft, 2019-05-09:
Hi Dan, thanks for your reply. Many schools use Techsoft 2D Designer to draw (and sometimes cut from depending on the machine) https://www.techsoft.co.uk/Products/Software/2DDesignV2 This program was developed in the late 90s and hasn't really been updated since then. Very frustrating for school student users as it doesn't even have an undo function. Therefore we thought we'd give designing a web based solution a go. Our dxf export (using your maker.js) will import into Techsoft (that's not really what we want to achieve) but won't import into LaserCut - Lasercut seems to be the Windows program that many Laser cutters are connected to. The frustrating thing is that if we import into Techsoft the Maker.js created dxf file and export it again - then it'll open in LaserCut. We don't have a Laser cutter or the Laser Cut software but I'm making enquiries into how to obtain it. I hope this info helps. Your assistance is much appreciated. 
Thanks Dave. 
p.s. if you'd like to see our 2D Designer App click here: https://www.focuselearning.co.uk/u/33147/gCzfinFftgftuscpvdADBaweFAqggdCmF

focusedsoft, 2019-05-09:
Hi Dan, we've found the download link for Lasercut software (see below). This link explains how to install it. There is a demo mode version that's free and we've downloaded it. Let me know if you need anything else. Kind regards, Dave.
https://smokeandmirrors.store/pages/installing-lasercut-5-3-windows-xp-vista-7

danmarshall, 2019-05-10:
Hi @focusedsoft, I was not able to run the Lasercut software because it says it requires a USB dongle.
At any rate, I have been looking at the DXF and one thing I have found was that Maker.js was not outputting the "CONTINUOUS" line style. So I added line styles to the output. Since I cannot run the Lasercut software, can you try this:

Copy the DXF code below and save it as a file with .dxf extension, then try to open it with your Lasercut software:
```
0
SECTION
2
HEADER
9
$INSUNITS
70
4
0
ENDSEC
0
SECTION
2
TABLES
0
TABLE
2
LTYPE
0
LTYPE
72
65
70
64
2
CONTINUOUS
3
______
73
0
40
0
0
ENDTAB
0
TABLE
2
LAYER
0
ENDTAB
0
ENDSEC
0
SECTION
2
ENTITIES
0
CIRCLE
8
0
10
0
20
0
40
90
0
CIRCLE
8
0
10
25
20
25
40
10
0
ARC
8
0
10
0
20
0
40
50
50
225
51
315
0
LINE
8
0
10
-35
20
20
11
-15
21
20
0
ENDSEC
0
EOF
```

If it works for you, I'll send a PR right away. [Here's the updated code](https://github.com/microsoft/maker.js/compare/master...danmarshall:dxf-linetype)

focusedsoft, 2019-05-10:
Hi Dan, I will ask Pete to check out the new dxf export this morning and get back to you.

focusedsoft, 2019-05-10:
Dan, FYI: The Lasercut installation file does install a demo version as well as the "dongle version" By default it installs into this folder C:/LaserCut53.  and can be run from: Lasercut53_Demo.exe 
regards, Dave

focusedsoft, 2019-05-10:
Dan, Great news, the dxf file I've created from your earlier post will load into Lasercut! 
For some unknown reason (to me as I'm not a programmer) I actually had to import it twice (without any changes) as Lasercut identified the image initially as being 'too big'.
Please view this video to see what I mean: http://www.focusdownload.com/camtasia/maker.js-import-into-lasercut/maker.js-import-into-lasercut.html
One more question, if you don't mind me asking: 
Our maker.js dxf export from focuselearning site does not export any colors which have been allocated to lines. This is important as lasers need to know what type of cut to make and that's allocated to different line colors. Is there anything we can tweak to achieve this or is it something that could be changed in your maker.js code? Many thanks for your help on all this. regards, Dave

danmarshall, 2019-05-10:
Hi Dave, you can color by layer. See the layer doc at https://maker.js.org/docs/intermediate-drawing/#Layers and this issue: https://github.com/microsoft/maker.js/issues/393

pete273, 2019-05-10:
Hi Dan. Thanks for your help. The updated DXF seems to open OK in the laser cutter software.
Can you send a link where we can download an updated maker.js file for our web site.

danmarshall, 2019-05-13:
HI @pete273 - I've added these changes in the latest build 0.14.0, thanks for your help! Also, your 2D Designer App is really cool, will that be publicly accessible?

PodgPow, 2019-05-13:
layer name & layer colour in DXF export #393 

I changed the library to export to DXF. Now setting the color and name of the layer is done very simply.
(And how to build a project-I don't know - errors (issue #405))

Example:
`Dots = new makerjs.models.ConnectTheDots("Array with coordinates");`
`Dots.layer = ["Name layer", "color layer"];`

How can I do the same for `ConnectTheDots` thing using the description in [https://maker.js.org/docs/intermediate-drawing/#Layers ](url)?

pete273, 2019-05-13:
> HI @pete273 - I've added these changes in the latest build 0.14.0, thanks for your help! Also, your 2D Designer App is really cool, will that be publicly accessible?

No, the whole site (https://www.focuselearning.co.uk) is access via subscription

closed 2019-05-13 by pete273

z3dev, 2019-05-13:
Just adding my two cents... DXF is a pretty over used and abused format. There are so many old and new applications generating files in different versions of DXF, with different contents to represent the same shape, etc. The problem is ... there’s no standard.

So, comparing files from Techsoft 2D design versus Autocad 2018 is like comparing cars of 1910 with today. There’s not much the same anymore. Maker.js has to find the middle ground, which means exporting with the headers required for the target version of Autocad.

danmarshall, 2019-05-14:
@z3dev agree - DXF is like a dark art at this point. Would you mind trying out a couple designs in the [demos](https://maker.js.org/demos/#content) to see if their DXF output is readable in your apps?

z3dev, 2019-05-19:
@danmarshall I exported heart, spokes, exo panel, and dogbone demonstrations to DXF, and imported into JSCAD. There were no issues with reading the files, with all the path segments converted.

The conversions highlighed a conversion issue with the LINE entities, and I raised a new JSCAD IO issue.

It does seem that order of the entities are quite mixed. I'm not sure the order of the segments could be discerned easily by reading the DXF file. Can the entities be generated in the same fasion as the JSCAD file, which seems to have an order?

PodgPow, 2019-05-29:
How to output blocks with primitives when exporting to DXF?