microsoft/maker.js #405: Build maker.js

PodgPow, 2019-05-07:


PodgPow, 2019-05-07:
**npm install
npm run build**
../../../../../node_modules/@types/opentype.js/index.d.ts(34,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'. 
Name 'CanvasRenderingContext2D' is available in this file (index.d.ts) - I do not have a reason !?

danmarshall, 2019-05-07:
Hi @PodgPow - Looks like the [opentype.js type definitions](https://github.com/DefinitelyTyped/DefinitelyTyped/blob/master/types/opentype.js/index.d.ts) have been updated, and locked into the "dom" lib. Maker.js uses these without dom.

Short term fix: I should lock down to a specific version to get you unblocked. PR coming soon.
Long-term fix: I will send a PR to update the types on DefinitelyTyped to not only use the "dom" lib.

danmarshall, 2019-05-08:
@PodgPow - I did a fresh install on a new machine and it built for me without error. Can you copy/paste more details on your error? 
Here is an example from my shell:
```bash
$npm run build

> makerjs-dev@1.0.0 build /mnt/c/src/maker.js
> lerna run build

lerna notice cli v3.13.4
lerna info Executing command in 3 packages: "npm run build"
lerna info run Ran npm script 'build' in 'fonts' in 3.3s:

> fonts@1.0.0 build /mnt/c/src/maker.js/packages/fonts
> tsc -p .

lerna info run Ran npm script 'build' in 'makerjs' in 17.3s:

> makerjs@0.13.1 prebuild /mnt/c/src/maker.js/packages/maker.js
> npm run gen-cascades

> makerjs@0.13.1 gen-cascades /mnt/c/src/maker.js/packages/maker.js
> npm run project && node ./target/cascadable.js > ./src/core/cascades.ts

> makerjs@0.13.1 project /mnt/c/src/maker.js/packages/maker.js
> typedoc ./src --json ./dist/project.json

Using TypeScript 2.7.2 from /mnt/c/src/maker.js/node_modules/typedoc/node_modules/typescript/lib

JSON written to /mnt/c/src/maker.js/packages/maker.js/dist/project.json

> makerjs@0.13.1 build /mnt/c/src/maker.js/packages/maker.js
> tsc -p target && node ./target/build.js

lerna info run Ran npm script 'build' in 'playground' in 7.3s:

> playground@1.0.0 build /mnt/c/src/maker.js/packages/playground
> tsc -p . && tsc -p src/worker

lerna success run Ran npm script 'build' in 3 packages in 24.6s:
lerna success - fonts
lerna success - makerjs
lerna success - playground
$
```

PodgPow, 2019-05-08:
### Thanks for the reply
I'm trying to build in Windows 10.
Maybe I do something wrong (I have a little experience in developing on JS, I have experience only with C++, so sorry). In addition, I am busy with my project and understand in detail-no time

First I installed:
`npm install opentype.js`

`> npm run build`
`> makerjs-dev@1.0.0 build MYFOLDERS\maker.js`
`lerna run build`

`lerna notice cli v3.13.4
lerna info Executing command in 3 packages: "npm run build"
lerna ERR! npm run build exited 2 in 'fonts'
lerna ERR! npm run build stdout:`

`> fonts@1.0.0 build c:\Users\IOleg\Documents\MyGit\maker.js\packages\fonts`
`> tsc -p .`
`../../../../../node_modules/@types/opentype.js/index.d.ts(34,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'.`
`...........`
`lerna ERR! npm run build stderr:`
`npm ERR! code ELIFECYCLE`
`npm ERR! errno 2`
`npm ERR! fonts@1.0.0 build: 'tsc -p .`  - **What is this?**
`npm ERR! Exit status 2`
`npm ERR!`
`npm ERR! Failed at the fonts@1.0.0 build script.`
`npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`
`npm WARN Local package.json exists, but node_modules missing, did you mean to install?`
`.......`
`lerna ERR! npm run build exited 2 in 'fonts'`
`lerna WARN complete Waiting for 1 child process to exit. CTRL-C to exit immediately.`
`npm ERR! code ELIFECYCLE`
`npm ERR! errno 2`
`npm ERR! makerjs-dev@1.0.0 build: 'lerna run build'`
`npm ERR! Exit status 2`
`npm ERR!`
`npm ERR! Failed at the makerjs-dev@1.0.0 build script.`
`npm ERR! This is probably not a problem with npm. There is likely additional logging output above.`

danmarshall, 2019-05-08:
Hi @PodgPow - can you tell me more about your scenario - I'm not sure that you actually need to "build" Maker.js unless you are modifying the core code. I'm happy to help you walk through setting up a project that uses Maker.js if you need it.

closed 2021-07-17 by danmarshall