microsoft/maker.js #407: How to calculate the area?

doox911, 2019-05-13:
Hello!) How to calculate the area in arch or or another figures?

danmarshall, 2019-05-13:
Hello, I wasn't sure how many people would want/need area calculation, but my thought was to implement something similar to the algorithm here: https://alienryderflex.com/polygon_area/

doox911, 2019-05-13:
It would be great if you added this.

MontassarAbdelhedi, 2023-02-14:
Hello, is the area feature is introduced in makerjs yet? if not, can you please write a sample code that returns the area of a rectangle? (not length*width)

Thank you

danmarshall, 2023-02-14:
@MontassarAbdelhedi it's not planned and not requested as much as other features. Please check my comment above for a cool algorithm. If your rectangle is not rotated, try [makerjs.measure.modelExtents](https://maker.js.org/docs/api/modules/makerjs.measure.html#modelextents) which will give rectangular outer extents.