microsoft/maker.js #409: SVG export: width and Height header to millimeter

ericLugtigheid, 2019-05-27:
Due to dpi settings at Easel (inventables) the scaling of the SVG filesturnes out wrong.

It would be great if we can set (by default?) the SVG export: width and Height header to millimeter.

So: 
'<svg **width="360" height="160"** viewBox="0 0 360 160" xmlns="http://www.w3.org/2000/svg">'
would become 
'<svg **width="360mm" height="160mm"** viewBox="0 0 360 160" xmlns="http://www.w3.org/2000/svg">'

This way the scaling comes out right.

danmarshall, 2019-05-28:
Hi @ericLugtigheid , you can just set your model's units to 'mm' to make this work. See https://maker.js.org/docs/basic-drawing/#Units in the docs.
Also see https://maker.js.org/playground/?script=kerf-compensation for an example, and export it as svg.

ericLugtigheid, 2019-05-28:
Hi Dan,

That's a very simple yet powerful feature, didn't know that. Thank you!
I'll study the rest as well!

Kind regards,
Eric Lugtigheid

danmarshall, 2019-05-28:
Thanks Eric! Please let me know how the docs could have been improved to have better exposed this.

closed 2019-08-09 by danmarshall