microsoft/maker.js #410: DXF export not including line colours

focusedsoft, 2019-05-29:
Hi Dan, we've been making good progress on the export of dxf files from our App. (https://github.com/microsoft/maker.js/issues/404#)
However, when a colour has been allocated to a shape with a curve, the colour info for curves isn't export. i.e. it doesn't display in Lasercut. We think this is in the maker code rather than ours and would welcome any suggestions:  Here's a demo:
http://www.focusdownload.com/camtasia/dxf-colour-info-for-curves-not-exporting/dxf-colour-info-for-curves-not-exporting.html
Let me know if you need any further info. Thanks, Dave.
[image]

This is the dxf file - it should display 2 rounded corner rectangles with red lines all round:
0
SECTION
2
HEADER
9
$INSUNITS
70

0
ENDSEC
0
SECTION
2
TABLES
0
TABLE
2
LTYPE
0
LTYPE
72
65
70
64
2
CONTINUOUS
3
______
73
0
40
0
0
ENDTAB
0
TABLE
2
LAYER
0
LAYER
2
red
70
0
62
1
6
CONTINUOUS
0
ENDTAB
0
ENDSEC
0
SECTION
2
ENTITIES
0
LINE
8
red
10
125
20
-109
11
125
21
-70
0
LINE
8
red
10
130
20
-65
11
205
21
-65
0
LINE
8
red
10
210
20
-70
11
210
21
-109
0
LINE
8
red
10
205
20
-114
11
130
21
-114
0
ARC
8
0
10
130
20
-109
40
5
50
180
51
270
0
ARC
8
0
10
130
20
-70
40
5
50
90
51
180
0
ARC
8
0
10
205
20
-70
40
5
50
0
51
90
0
ARC
8
0
10
205
20
-109
40
5
50
270
51
360
0
LINE
8
red
10
170
20
-132
11
170
21
-96
0
LINE
8
red
10
175
20
-91
11
236
21
-91
0
LINE
8
red
10
241
20
-96
11
241
21
-132
0
LINE
8
red
10
236
20
-137
11
175
21
-137
0
ARC
8
0
10
175
20
-132
40
5
50
180
51
270
0
ARC
8
0
10
175
20
-96
40
5
50
90
51
180
0
ARC
8
0
10
236
20
-96
40
5
50
0
51
90
0
ARC
8
0
10
236
20
-132
40
5
50
270
51
360
0
ENDSEC
0
EOF

danmarshall, 2019-05-29:
Thanks Dave. Can you share the JSON of the model before it gets passed to the `makerjs.exporter.toDXF()` function call?

focusedsoft, 2019-05-30:
Hi Dan, here's a screen capture. A line path has the "layer attribute: red"  but the Arc path does not.
I hope this help. regards, Dave
[image]

danmarshall, 2019-05-30:
Hi Dave, I think you've answered your own question: the arc does not have the `layer:"red"` attribute. I think your program is generating this. Optionally, you may want to add the attribute at the model level, instead of each individual path element.

focusedsoft, 2019-05-30:
Hi Dan, all sorted, very many thanks. We've added the attribute at model level and it works a treat. Not being a "coder" I get to ask the "stupid" questions (AKA being the ideas man) and most of the time, if I'm being honest, I'm not actually sure what I'm asking  ;-)  But you do tend to get to the bottom of things a lot quicker that way. All the best, Dave.

danmarshall, 2019-05-30:
No worries, glad you’re up and running !

closed 2019-05-30 by danmarshall