microsoft/maker.js #411: export the blocks to DXF (BLOCKS section of the DXF file)

PodgPow, 2019-05-29:
How to output blocks with primitives when exporting to DXF?

danmarshall, 2019-05-29:
Hi @PodgPow , I may also be needing to have blocks exported, when #133 is fully implemented. Can you add details about your use case?

PodgPow, 2019-05-30:
I want to use blocks for exporting to the DXF path, consisting of several circuits (in SVG everything is fine with this).

danmarshall, 2019-05-30:
Can you say what advantage it is to have things in blocks?

PodgPow, 2019-05-30:
One example - is the transfer to a DXF shape with a hole, so that you can work in AutoCAD (ArtCAM) with the shape as a solid (not fragmented by contour) object

cgowen, 2021-03-23:
I would like to use makerjs to produce parametric dxf AAMA files for CNC textile cutters. (Lectra Gerber etc)
Some of the software used by these machines requires dxf files with blocks to separate the patterns from one another.

There are other options where each pattern could be exported as it's own dxf file but I'd prefer to keep it all together if possible so that I can use the file to reference the parts together.

It looks like all the magic is happening in the packages/maker.js/src/core/dxf.ts file. I've downloaded the Autodesk DXF reference but honestly, it's pretty overwhelming.
It looks like there needs to be a blocks section added to the exported file and then the block references can be included as an entity to inside the drawing with it's own origin?

Is there a simpler reference that I'm missing somewhere?

Thanks

danmarshall, 2021-03-24:
HI @cgowen , you're not missing anything - the DXF format is not super easy.

cgowen, 2021-03-26:
I've done a small amount of digging through Audodesk references and files I exported with the exdxf library and have come up with a short breakdown of how to add blocks to a dxf file. I still don't know enough javascript to make this happen. 
I'm hoping that having the file structure may be helpful to someone who might be a better coder than I am.

https://anonfiles.com/w5DaHbm2uc/DXF_Format_Cheatsheet_pdf
https://anonfiles.com/u8D7H4mfu7/handmade_dxf