microsoft/maker.js #412: rounding slow?

blackmad, 2019-06-06:
Hi,

I've been loving maker.js for some generative laser cutting work. I've been trying to understand how to make it faster and something I've noticed is that by far, the round() call stands out in my performance debugging in chrome. 

I looked and noticed that the code is doing a lot of string manipulation. I replaced the first line of it with

        if (accuracy === void 0) { return (n * 1000000) / 1000000 }

and it seems like it cut the running time of my code in half. Is there a major reason why this solution wouldn't work for all cases?

blackmad, 2019-06-06:
well, the answer is "because it doesn't work" - it definitely breaks some calculations

closed 2019-06-06 by blackmad

danmarshall, 2019-06-07:
Hi David, thanks for the feedback. Yes the round() function is called the most and should be optimized further. I wasn’t thrilled to use strings, but it works to get the desired result most of the time. I’m happy to take a PR. One thing I would like is a lot more unit tests around it.

One thing you may try is to overwrite makerjs.round with your own function in isolated cases, then revert it afterwards.

Also I hope there may be other areas in your code that might be optimized. If you are using the Playground for example, it definitely can be inefficient because it’s running the code fresh each time a parameter changes. In your own code, you can definitely apply a lot of caching of things you know won’t need recalculation.

blackmad, 2019-06-07:
Hey Dan, thanks for the reply.

Do you have an intuition why a Math.round scaling solution doesn't work?
What lead you to this implementation?

Is makerjs.round mainly being called for approximate floating point
near-equality? In that case, why not have a nearEquals function that checks
"Math.abs(a -b ) < epsilon" which would be a lot less work than using
string?

danmarshall, 2019-06-07:
There's a few places that should be optimized for the near equality using the technique you mentioned, instead of rounding.

The string-based implementation was added in #254 to provide expected results. See the discussion at https://stackoverflow.com/questions/11832914/round-to-at-most-2-decimal-places-only-if-necessary and note that the accepted answer is not correct for many examples. Specifically see the case for the number 1.005, which should be rounded to 1.01 when you want 2 decimal places.