microsoft/maker.js #415: DXF import fails into AutoCAD for captions

boomer196, 2019-07-01:
I would appreciate some help...

I am getting an error while importing DXF that was generated from a model that contains captions.  I tried both AutoCAD 2018 and AutoCAD 2019.

I used the example from the docs:
```
var makerjs = require('makerjs');
var square = new makerjs.models.Square(100);
square.caption = {
    text: "a square",
    anchor: new makerjs.paths.Line([0, 50], [100, 50])
};
var dxf = makerjs.exporter.toDXF(square);
```

The DXF generated is:
```
0
SECTION
2
HEADER
9
$INSUNITS
70
4
0
ENDSEC
0
SECTION
2
TABLES
0
TABLE
2
LTYPE
0
LTYPE
72
65
70
64
2
CONTINUOUS
3
______
73
0
40
0
0
ENDTAB
0
TABLE
2
LAYER
0
ENDTAB
0
ENDSEC
0
SECTION
2
ENTITIES
0
LINE
8
0
10
0
20
0
11
100
21
0
0
LINE
8
0
10
100
20
0
11
100
21
100
0
LINE
8
0
10
100
20
100
11
0
21
100
0
LINE
8
0
10
0
20
100
11
0
21
0
0
MTEXT
10
50
20
50
40
9
71
5
72
1
1
a square
50
0
0
ENDSEC
0
EOF
```

The error is:
```
Undefined group code 10 for object on line 98.
Invalid or incomplete DXF input -- drawing discarded.
```
Lines 96-98 for context are:
```
MTEXT
10
50
```

Here is the JSON:
```
{
  "models": {
    "square": {
      "paths": {
        "ShapeLine1": {
          "type": "line",
          "origin": [
            0,
            0
          ],
          "end": [
            100,
            0
          ]
        },
        "ShapeLine2": {
          "type": "line",
          "origin": [
            100,
            0
          ],
          "end": [
            100,
            100
          ]
        },
        "ShapeLine3": {
          "type": "line",
          "origin": [
            100,
            100
          ],
          "end": [
            0,
            100
          ]
        },
        "ShapeLine4": {
          "type": "line",
          "origin": [
            0,
            100
          ],
          "end": [
            0,
            0
          ]
        }
      },
      "caption": {
        "text": "a square",
        "anchor": {
          "type": "line",
          "origin": [
            0,
            50
          ],
          "end": [
            100,
            50
          ]
        }
      }
    }
  },
  "origin": [
    0,
    0
  ]
}
```

If you remove the `caption` from the model, then the generated DXF works just fine.
```
0
SECTION
2
HEADER
9
$INSUNITS
70
4
0
ENDSEC
0
SECTION
2
TABLES
0
TABLE
2
LTYPE
0
LTYPE
72
65
70
64
2
CONTINUOUS
3
______
73
0
40
0
0
ENDTAB
0
TABLE
2
LAYER
0
ENDTAB
0
ENDSEC
0
SECTION
2
ENTITIES
0
LINE
8
0
10
0
20
0
11
100
21
0
0
LINE
8
0
10
100
20
0
11
100
21
100
0
LINE
8
0
10
100
20
100
11
0
21
100
0
LINE
8
0
10
0
20
100
11
0
21
0
0
ENDSEC
0
EOF
```

danmarshall, 2019-07-08:
Thanks for filing this. I've found this thread on the AutoCAD forum:
https://forums.autodesk.com/t5/autocad-forum/opening-dxf-file-with-mtext-error-undefined-group-code-10-for/td-p/2855068
which states:
> AutoCAD doesn't support entities only (i.e. only has an ENTITIES section) dxf files with entity types that were introduced after Acad R12.  We actually have two completely different code paths for dealing with R12 and earlier dxf files and R13 and newer files.  When we encounter an entities only dxf file, we use the R12 dxf code path, so it only supports entity types that were present in the R12 dxf format.

It seems that the fix might be to:
>  include a header section to let AutoCAD know what version the file is.

boomer196, 2019-07-31:
Thanks!  I added the version to the header, but once you do that it wants an `APPID` as well.  Does Maker.js support that?  If not, I'll try to figure it out, but wanted to ask that first.

danmarshall, 2019-07-31:
Hmm. Can you use "Maker.js" as the APPID ? Or is it in a specific format?

boomer196, 2019-07-31:
It's a specific format.  It's a `TABLES` section.  

https://knowledge.autodesk.com/search-result/caas/CloudHelp/cloudhelp/2018/ENU/AutoCAD-DXF/files/GUID-6E3140E9-E560-4C77-904E-480382F0553E-htm.html

danmarshall, 2019-07-31:
So maybe add another table [here](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/dxf.ts#L396) ?

danmarshall, 2019-07-31:
Off the top of my head:
```
        function tables() {
            append("2", "TABLES");

            table(lineTypesOut);
            table(layersOut);
            table(appIdOut);
        }

        function appIdOut() {
            append("0", "APPID",
                "100",
                "AcDbRegAppTableRecord",
                "2",
                "Maker.js",
                "70",
                0
            );
        }
```

closed 2019-08-09 by boomer196