microsoft/maker.js #416: Grouping Paths

fobus42, 2019-07-05:
Hello,

We are using makerjs in our project. We are importing SVG paths combining together and exporting as DXF. But we want to group each svg path to a group.

Currently it is exporting all lines, circles etc.. in SVG path as an separate element. Can we export DXF with AutoCad like grouping function?

Thanks

danmarshall, 2019-07-08:
Hi @fobus42 - as for the "AutoCad like grouping function" - can you reverse engineer how this works in a DXF file? It may be BLOCKS, which is a related feature request #411

z3dev, 2021-03-23:
Blocks are basically a group of DXF entities, I.e  lines, shapes, textures, etc.

The advantage of blocks is that the group of entities can be reused several times in the document. For example, placing a ‘bush’ at several different locations in a document.

There’s no definition of groups in DXF.