microsoft/maker.js #423: gcode

ericLugtigheid, 2019-08-05:
Has anyone ever tried or implemented GCODE as output for CNC routers?

danmarshall, 2019-08-05:
Hi @ericLugtigheid , check out http://jscut.org/ or the [gcode topic on GitHub](https://github.com/topics/gcode)

ericLugtigheid, 2019-08-17:
Hi Dan,
Thanks for your idea. I already checked jscut, very neat indeed but i would like to have a direct output in gcode.... I'll check out the topic on GitHub.

Eric

danmarshall, 2019-08-18:
My personal experience with gcode has been with 3D printers. The first commands were machine-specific codes to initialize the machine. The next commands were to position the part relative to the machine’s working area. Finally the remaining commands were the actual part. But these moves also have optimizations. In 3D printing, overlap is ok and the main optimization is for time. With a CNC router however, you may have some z commands along the way which leave tabs that you later break away, so the part is held in place securely.
So the bottom line is, IMO, that the gcode is going to be very specific for a machine and a given scenario. You’d need to be able to tweak a lot of parameters to get your desired result.
Can you give any specific details about the gcode you seek? Is there some generic gcode “move commands” that you would find useful? How would you use the gcode?

rmm5t, 2019-08-20:
> With a CNC router however, you may have some z commands along the way which leave tabs that you later break away, so the part is held in place securely.

To add to the complications of this, gcode must also take into consideration the kerf of the cut. e.g. the diameter of a router bit or even the kerf of a laser.

msurguy, 2019-09-24:
@ericLugtigheid check out this project : https://github.com/makercam/openjscam

danmarshall, 2019-09-25:
@ericLugtigheid - I just found a few more gcode projects:
* https://github.com/oneislandearth/gcode
* https://github.com/drandrewthomas/gcodercnc2d5

danmarshall, 2019-09-25:
Also wanted to point out, from looking at this sample code at https://github.com/oneislandearth/gcode :
```
// Start the spindle
code.startSpindle();

// Start the coolant
code.startCoolant();

// Drop the mill to the bottom of the board
code.dropMill();

// Cut lengthways 65.5cm
code.feedLinear([0, 65.5]);
```
Operations such as `startSpindle` and `startCoolant` make me believe that gcode is out of scope for Maker.js - I'm going to close this issue. I hope you've found enough resources in the links above to make it work. Thanks, Dan

closed 2019-09-25 by danmarshall