microsoft/maker.js #428: Can't Draw Hatch's and Patterns

Monarkos, 2019-08-08:
Hi there,

I'm using the lib to draw on some section based on dxf file, but for no reason the Hatch Pattern is ignored during de conversion to polylines, then those parts are aways missing.
Could anyone help me?

Thank you

danmarshall, 2019-08-08:
Hello, can you provide a code sample?

Monarkos, 2019-08-09:
Hey @danmarshall, thank you here it is.

` let file = new File([ng.binary], ng.binary.type, metadata);
           var reader = new FileReader();
           reader.onload = function(e: any) {
               ng.makerJsModel = null;
               var dxfContents = e.target.result;
               var helper = new dxf.Helper(dxfContents);
               let filter = {
                   models: {}
               }
               debugger
               let polylines = helper.toPolylines().polylines;
               if (polylines.length === 0) {
                   configuration.addQuantityMovements();
                   resolve(null);
               } else {
                   polylines.forEach((line, i) => {
                       // filter.models['filter' + i] = new makerjs.models.ConnectTheDots(closed, line.vertices);
                       filter.models['filter' + ((Math.random() * 10) + 1).toString()] = new makerjs.models.ConnectTheDots(closed, line.vertices);
                   });
                   let object = {
                       dxfImage: filter,
                       imageId: ng.id,
                       image: ng.image
                   };
                   resolve(object);
               }
           };
           reader.readAsBinaryString(file);
       })`

danmarshall, 2019-08-09:
It appears the your `dxf.Helper` does not emit the hatch pattern as a polyline. Can you tell me what library you are using for the `dxf`?

Monarkos, 2019-08-09:
Hi @danmarshall I'm using [](https://github.com/bjnortier/dxf)

Monarkos, 2019-08-09:
Do

> It appears the your `dxf.Helper` does not emit the hatch pattern as a polyline. Can you tell me what library you are using for the `dxf`?

Can you recommend some one that works for me?

danmarshall, 2019-08-09:
Thanks. Can you also provide a link to a sample dxf file with the hatch issue?

Monarkos, 2019-08-09:
Yes, there is a model here [DXF File](https://we.tl/t-TBNT8T6Mnn)

ru1973, 2019-08-10:
@danmarshall - I think what Monarkos might be referring to is that you cannot create polylines from some files that are created using maker.js - specifically, any file that uses arcs. I can create DXF files of maker.js demos that only use lines, import into Draftsight CAD, and create polylines from the line data. When I do the same for any maker.js demo that contains arcs ( and or can contain lines too ) I cannot create polylines from the data. I documented the demo example makerjs-smile, and used the default  values for the DXF file. When I import into Draftsight, the endpoints of the arcs that should be coincident are slightly off. I am going to guess that there is a rounding error in a function, and it might be going from radians to angle or the opposite. Just a guess. I dimensioned to 8 places, and the error shows after 4 places. See PDF documentation.

I am pretty sure the error occurs because DXF code to represent arcs defines the center point of the arc, then it defines start angle and end angle. If the code for those angles is slightly off, the endpoints of any adjacent lines will mot be coincident. 

Ralph
[makerjs-smile.pdf](https://github.com/microsoft/maker.js/files/3489154/makerjs-smile.pdf)

ru1973, 2019-08-10:
@danmarshall - I think what Monarkos might be referring to is that you cannot create polylines from some files that are created using maker.js - specifically, any file that uses arcs. I can create DXF files of maker.js demos that only use lines, import into Draftsight CAD, and create polylines from the line data. When I do the same for any maker.js demo that contains arcs ( and or can contain lines too ) I cannot create polylines from the data. I documented the demo example makerjs-smile, and used the default  values for the DXF file. When I import into Draftsight, the endpoints of the arcs that should be coincident are slightly off. I am going to guess that there is a rounding error in a function, and it might be going from radians to degrees or the opposite. Just a guess. I dimensioned to 8 places, and the error shows after 4 places. See PDF documentation.

I am pretty sure the error occurs because DXF code to represent arcs defines the center point of the arc, then it defines start angle and end angle. If the code for those angles is slightly off, the endpoints of any adjacent lines will not be coincident. 

Ralph
[makerjs-smile.pdf](https://github.com/microsoft/maker.js/files/3489189/makerjs-smile.pdf)

------------------
edit 1: When I tested again with Accuracy to MAX and did not use polylines, I still could not convert coincident arcs to polylines in Draftsight. But, when I output DXF with default Accuracy and checked Use POLYLINES, the smile is a Polyline in Draftsight.

edit 2: Output DXF for the same demo with Accuracy at MAX and checked Use POLYLINES. Imported into Draftsight, exploded Polyline for smile, exported DXF, and looked at the code. All of the math is slightly off including center points and start and end angles.

edit 3: Code snippet in \maker.js-master\docs\playground\index.html

`                          <option value=".0001" selected>.0001</option>   `

Changing the default option from ".0001" to 15 decimal places ( ".000000000000001" ) might fix it, although it may break something somewhere else. When I output a DXF release 12 from Draftsight of smile, it formatted the centerpoints of the arcs to 14-16 places.

Monarkos, 2019-08-12:
Hi, @ru1973, that is not the case of my problem, was exactly what @danmarshall was talking before, I'm not even trying to create any file, I'm drawing on the page from a DXF file that match the specifications, but I can't figure out why the hatch part of the DXF is being ignored and than those parts of my drawing are missed. I guess @danmarshall was correctly and the problem is my library that is not drawing polylines for hatch, and my question was how to fix/work around that issue?

ru1973, 2019-08-12:
@Monarkos - sorry for the confusion. But, when I look at your DXF provided file, I see 2 hatch patterns within separate blocks. They are somewhat hard to see with a white background, and their density might make it harder to see. Did you have more than 2 hatch patterns?

[Hatch_Pattern.pdf](https://github.com/microsoft/maker.js/files/3493311/Hatch_Pattern.pdf)

Monarkos, 2019-08-12:
@ru1973 On that example was just those two, and you are correct, they are a little hard to catch depending on the background, but, was just an example, i already tried to change the patter to full solid and they still don't appear on the drawing =(
Most likely they are being ignored on transformation to polylines...
I just have no clue how to fix that...
Did worked for you?

ru1973, 2019-08-12:
@Monarkos - the other thing you could look at is what the value of FILLMODE is set to in DXF - DXF code snippet below

$FILLMODE
 70
     1

The "1" defines that the hatching will display. It was set to 1 in your example file. Perhaps some other files have the value set to "0" - which defines that the hatching will not display.

Monarkos, 2019-08-12:
@ru1973 I will see that ether, but even on that example file the hatch is not appearing.. =(

ru1973, 2019-08-12:
@Monarkos - I opened your provided DXF in Draftsight, edited the BLOCK FILTRO FINO 592 x L-292, then did a HATCH edit, and then merely selected OK. It made a denser hatch pattern. Is it possible that some of your DXF files are initially created with imperial units verses metric?  See DXF codes for Hatch below.

[image]