microsoft/maker.js #435: Shape

Neat-1, 2019-08-22:
Hi 
I am trying to make this shape with offset on inside & parameters Any help or direction would be appreciated.
Left size
Base size
Right size
Diagonal 1 
Diagonal 2
[image]

rmm5t, 2019-08-22:
I'd suggest taking questions like this to a site like [StackOverflow](http://stackoverflow.com/). GitHub Issues are best reserved for discussing feature requests and bug reports. 

Meanwhile, please read through the docs, and play in the playground:

- https://maker.js.org/docs/basic-drawing/
- https://maker.js.org/playground/?script=rimbox3d

danmarshall, 2019-08-22:
Appreciate the support, @rmm5t ! 
For background on this: I've specifically asked @Neat-1 to post this here. Currently this community has very little traction on StackOverflow. There's few questions, and fewer answerers. https://stackoverflow.com/search?q=makerjs

I don't mind (ab)using GitHub issues in the interim, to keep the knowledge central.

Neat-1, 2019-08-29:
Hi all
I have a better understanding of how Maker works but I am still stuck ,in the picture I can get the intersections with arc or circle but I am not sure how to get the lines to link to these points.
Any help would be greatly appreciated.
[image]

Chris

Neat-1, 2019-08-29:
This is from Maker with two circles for intersection hope it makes sense
Chris
[image]

danmarshall, 2019-08-29:
That's the right start. I converted them to arcs, then call [makerjs.path.intersection](https://maker.js.org/docs/intermediate-drawing/#Intersection):
```
var makerjs = require('makerjs');

function shape(bottom, left, diag) {
  this.units = makerjs.unitType.Millimeter;
  this.paths = {
      bottom: new makerjs.paths.Line([0, 0], [bottom, 0]),
      leftArc: new makerjs.paths.Arc([0, 0], left, 0, 180),
      diagArc: new makerjs.paths.Arc([bottom, 0], diag, 90, 180)
  };
  var int = makerjs.path.intersection(this.paths.leftArc, this.paths.diagArc);
  console.log(int);
  this.paths.left = new makerjs.paths.Line([0,0], int.intersectionPoints[0]);
}

shape.metaParameters = [
      { title: "bottom", type: "range", min: 10, max: 3000, value: 2169.7470 },
      { title: "left", type: "range", min: 10, max: 3000, value: 967.3320 },
      { title: "diag", type: "range", min: 10, max: 3000, value: 1646.8345 }
];

module.exports = shape;

```

[image]

danmarshall, 2019-08-29:
Do the same thing on the right side:
```
var makerjs = require('makerjs');

function shape(bottom, left, right, diag1, diag2, showArcs) {
  this.units = makerjs.unitType.Millimeter;
  this.paths = {
      bottom: new makerjs.paths.Line([0, 0], [bottom, 0]),
      leftArc: new makerjs.paths.Arc([0, 0], left, 0, 180),
      rightArc: new makerjs.paths.Arc([bottom, 0], right, 0, 180),
      diagArc1: new makerjs.paths.Arc([bottom, 0], diag1, 135, 180),
      diagArc2: new makerjs.paths.Arc([0, 0], diag2, 0, 90)
  };
  var int1 = makerjs.path.intersection(this.paths.leftArc, this.paths.diagArc1);
  console.log(int1);
  this.paths.left = new makerjs.paths.Line([0,0], int1.intersectionPoints[0]);

    var int2 = makerjs.path.intersection(this.paths.rightArc, this.paths.diagArc2);
  console.log(int2);
  this.paths.right = new makerjs.paths.Line([bottom,0], int2.intersectionPoints[0]);

  this.paths.top = new makerjs.paths.Line(this.paths.left.end, this.paths.right.end);
  
  if (!showArcs) {
    delete this.paths.leftArc;
    delete this.paths.rightArc;
    delete this.paths.diagArc1;
    delete this.paths.diagArc2;
  }
  
}

shape.metaParameters = [
      { title: "bottom", type: "range", min: 10, max: 3000, value: 2169.7470 },
      { title: "left", type: "range", min: 10, max: 3000, value: 967.3320 },
      { title: "right", type: "range", min: 10, max: 3000, value: 848.0377 },
      { title: "diag1", type: "range", min: 10, max: 3000, value: 1646.8345 },
      { title: "diag2", type: "range", min: 10, max: 3000, value: 2748.0736 },
   { title: "show arcs", type: "bool", value: false }
];

module.exports = shape;
```
[image]

danmarshall, 2019-08-29:
To get the inner frame, call [makerjs.model.outline](https://maker.js.org/docs/api/modules/makerjs.model.html#outline) on the shape, passing a `true` for the `inside` parameter. I should add this to the [documentation](https://maker.js.org/docs/advanced-drawing/#Outlining%20a%20model).

Here's some code that does it:
```
var makerjs = require('makerjs');

function inFrame(distance, subtract, myShape) {
  
//JSON of a shape added for brevity
  myShape = myShape || {"paths":{"bottom":{"type":"line","origin":[0,0],"end":[2169.747,0]},"left":{"type":"line","origin":[0,0],"end":[-248.9036,934.761]},"right":{"type":"line","origin":[2169.747,0],"end":[2392,1352.8653]},"top":{"type":"line","origin":[-248.9036,934.761],"end":[2392,1352.8653]}},"origin":[0,0],"units":"mm"};

  this.models = { 
    myShape,
    inline: makerjs.model.outline(myShape, distance, 0, true)
  };
  
  var midTop = makerjs.point.middle(this.models["inline"].models["0"].paths["top_Bottom"]);
  var midBottom = makerjs.point.middle(this.models["inline"].models["0"].paths["bottom_Top"]);
  
  this.models.slot = new makerjs.models.Slot(midTop, midBottom, distance / 2);
  
  if (subtract) {
    makerjs.model.combineSubtraction(this.models.inline, this.models.slot);
  }
}

inFrame.metaParameters = [
      { title: "distance", type: "range", min: 10, max: 1000, value: 50 },
   { title: "subtract", type: "bool", value: true }
];

module.exports = inFrame;

```
Note I used the Playground to get the route of the paths on the inner frame, by clicking on the name of the top inner frame line:

[image]

I added a [slot](https://maker.js.org/playground/?script=Slot) at the midpoints, with a radius of half of the inset distance. 

[image]

Finally, I [subtracted](https://maker.js.org/docs/advanced-drawing/#Combining%20with%20Boolean%20operations) the slot from the inside outline.

[image]

Neat-1, 2019-08-30:
Hi Dan
This is a **Tremendous** help just what I required, and I am sure will help many others, I am looking and learning all the time.

Kind Regards
Chris

closed 2019-08-30 by danmarshall

Neat-1, 2019-09-01:
Hi Dan

I put all the code together and I have tried tried every combination I can think of but still can't get it to run.
They are all running when seperated? 

Thank you in advance

var makerjs = require('makerjs');

function shape(bottom, left, right, diag1, diag2, showArcs) {
    this.units = makerjs.unitType.Millimeter;

    this.paths = {
        bottom: new makerjs.paths.Line([0, 0], [bottom, 0]),
        leftArc: new makerjs.paths.Arc([0, 0], left, 0, 180),
        rightArc: new makerjs.paths.Arc([bottom, 0], right, 0, 180),
        diagArc1: new makerjs.paths.Arc([bottom, 0], diag1, 135, 180),
        diagArc2: new makerjs.paths.Arc([0, 0], diag2, 0, 90)
    };
    var int1 = makerjs.path.intersection(this.paths.leftArc, this.paths.diagArc1);
    console.log(int1);
    this.paths.left = new makerjs.paths.Line([0, 0], int1.intersectionPoints[0]);

    var int2 = makerjs.path.intersection(this.paths.rightArc, this.paths.diagArc2);
    console.log(int2);
    this.paths.right = new makerjs.paths.Line([bottom, 0], int2.intersectionPoints[0]);

    this.paths.top = new makerjs.paths.Line(this.paths.left.end, this.paths.right.end);

    //create a fillet
    var arc = makerjs.path.fillet(this.paths.left, this.paths.bottom, 5);
    var arc1 = makerjs.path.fillet(this.paths.right, this.paths.bottom, 5);
    var arc2 = makerjs.path.fillet(this.paths.right, this.paths.top, 5);
    var arc3 = makerjs.path.fillet(this.paths.left, this.paths.top, 5);

    //add the fillet to the model
    this.paths.arc = arc;
    this.paths.arc1 = arc1;
    this.paths.arc2 = arc2;
    this.paths.arc3 = arc3;

    //Show Arcs
    if (!showArcs) {
        delete this.paths.leftArc;
        delete this.paths.rightArc;
        delete this.paths.diagArc1;
        delete this.paths.diagArc2;
    }
    
  
}

function offset(shape,distance) {
    this.models = {shape,offset: makerjs.model.outline (shape, distance, 0, true)
                  }

     var midTop = makerjs.point.middle(shape.paths.top);
     var midBottom = makerjs.point.middle(shape.paths.bottom);
  
     this.models.slot = new makerjs.models.Slot(midTop, midBottom, distance / 2);
  
  if (subtract) {
    makerjs.model.combineSubtraction(this.models.offset, this.models.slot)
  
  }

 
    
offset.metaParameters = [
    { title: "bottom", type: "range", min: 300, max: 3000, value: 2000 },
    { title: "left", type: "range", min: 300, max: 900, value: 750 },
    { title: "right", type: "range", min: 300, max: 900, value: 750 },
    { title: "TLBR", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "TRBL", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "show arcs", type: "bool", value: false },
    { title: "distance", type: "range", min: 10, max: 1000, value: 50 },
    { title: "subtract", type: "bool", value: true }
];
  
module.exports = offset;