microsoft/maker.js #436: Code not running

Neat-1, 2019-09-01:
Hi Dan

I put all the code together and I have tried tried every combination I can think of but still can't get it to run.
They are all running ok when seperated? 

Thank you in advance

var makerjs = require('makerjs');

function shape(bottom, left, right, diag1, diag2, showArcs) {
    this.units = makerjs.unitType.Millimeter;

    this.paths = {
        bottom: new makerjs.paths.Line([0, 0], [bottom, 0]),
        leftArc: new makerjs.paths.Arc([0, 0], left, 0, 180),
        rightArc: new makerjs.paths.Arc([bottom, 0], right, 0, 180),
        diagArc1: new makerjs.paths.Arc([bottom, 0], diag1, 135, 180),
        diagArc2: new makerjs.paths.Arc([0, 0], diag2, 0, 90)
    };
    var int1 = makerjs.path.intersection(this.paths.leftArc, this.paths.diagArc1);
    console.log(int1);
    this.paths.left = new makerjs.paths.Line([0, 0], int1.intersectionPoints[0]);

    var int2 = makerjs.path.intersection(this.paths.rightArc, this.paths.diagArc2);
    console.log(int2);
    this.paths.right = new makerjs.paths.Line([bottom, 0], int2.intersectionPoints[0]);

    this.paths.top = new makerjs.paths.Line(this.paths.left.end, this.paths.right.end);

    //create a fillet
    var arc = makerjs.path.fillet(this.paths.left, this.paths.bottom, 5);
    var arc1 = makerjs.path.fillet(this.paths.right, this.paths.bottom, 5);
    var arc2 = makerjs.path.fillet(this.paths.right, this.paths.top, 5);
    var arc3 = makerjs.path.fillet(this.paths.left, this.paths.top, 5);

    //add the fillet to the model
    this.paths.arc = arc;
    this.paths.arc1 = arc1;
    this.paths.arc2 = arc2;
    this.paths.arc3 = arc3;

    //Show Arcs
    if (!showArcs) {
        delete this.paths.leftArc;
        delete this.paths.rightArc;
        delete this.paths.diagArc1;
        delete this.paths.diagArc2;
    }
    
  
}

function offset(shape,distance) {
    this.models = {shape,offset: makerjs.model.outline (shape, distance, 0, true)
                  }

     var midTop = makerjs.point.middle(shape.paths.top);
     var midBottom = makerjs.point.middle(shape.paths.bottom);
  
     this.models.slot = new makerjs.models.Slot(midTop, midBottom, distance / 2);
  
  if (subtract) {
    makerjs.model.combineSubtraction(this.models.offset, this.models.slot)
  
  }

 
    
offset.metaParameters = [
    { title: "bottom", type: "range", min: 300, max: 3000, value: 2000 },
    { title: "left", type: "range", min: 300, max: 900, value: 750 },
    { title: "right", type: "range", min: 300, max: 900, value: 750 },
    { title: "TLBR", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "TRBL", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "show arcs", type: "bool", value: false },
    { title: "distance", type: "range", min: 10, max: 1000, value: 50 },
    { title: "subtract", type: "bool", value: true }
];
  
module.exports = offset;

_Originally posted by @Neat-1 in https://github.com/microsoft/maker.js/issues/435#issuecomment-526925593_

danmarshall, 2019-09-08:
Hi @Neat-1 , Looks like you were missing a curly brace. I'd recommend using an offline editor like VSCode or something to catch syntax errors like this.

Next, I added a wrapper function to put it all together:

```
var makerjs = require('makerjs');

function shape(bottom, left, right, diag1, diag2, showArcs) {
    this.units = makerjs.unitType.Millimeter;

    this.paths = {
        bottom: new makerjs.paths.Line([0, 0], [bottom, 0]),
        leftArc: new makerjs.paths.Arc([0, 0], left, 0, 180),
        rightArc: new makerjs.paths.Arc([bottom, 0], right, 0, 180),
        diagArc1: new makerjs.paths.Arc([bottom, 0], diag1, 135, 180),
        diagArc2: new makerjs.paths.Arc([0, 0], diag2, 0, 90)
    };
    var int1 = makerjs.path.intersection(this.paths.leftArc, this.paths.diagArc1);
    console.log(int1);
    this.paths.left = new makerjs.paths.Line([0, 0], int1.intersectionPoints[0]);

    var int2 = makerjs.path.intersection(this.paths.rightArc, this.paths.diagArc2);
    console.log(int2);
    this.paths.right = new makerjs.paths.Line([bottom, 0], int2.intersectionPoints[0]);

    this.paths.top = new makerjs.paths.Line(this.paths.left.end, this.paths.right.end);

    //create a fillet
    var arc = makerjs.path.fillet(this.paths.left, this.paths.bottom, 5);
    var arc1 = makerjs.path.fillet(this.paths.right, this.paths.bottom, 5);
    var arc2 = makerjs.path.fillet(this.paths.right, this.paths.top, 5);
    var arc3 = makerjs.path.fillet(this.paths.left, this.paths.top, 5);

    //add the fillet to the model
    this.paths.arc = arc;
    this.paths.arc1 = arc1;
    this.paths.arc2 = arc2;
    this.paths.arc3 = arc3;

    //Show Arcs
    if (!showArcs) {
        delete this.paths.leftArc;
        delete this.paths.rightArc;
        delete this.paths.diagArc1;
        delete this.paths.diagArc2;
    }
}

function offset(shape, distance, subtract) {
    this.models = {
        shape, offset: makerjs.model.outline(shape, distance, 0, true)
    }

    var midTop = makerjs.point.middle(shape.paths.top);
    var midBottom = makerjs.point.middle(shape.paths.bottom);

    this.models.slot = new makerjs.models.Slot(midTop, midBottom, distance / 2);
    if (subtract) {
        makerjs.model.combineSubtraction(this.models.offset, this.models.slot)

    }
}

function all(bottom, left, right, tlbr, trbl, showArcs, distance, subtract) {
    var s = new shape(bottom, left, right, tlbr, trbl, showArcs);
    var o = new offset(s, distance, subtract);
    this.models = { s, o };

}

all.metaParameters = [
    { title: "bottom", type: "range", min: 300, max: 3000, value: 2000 },
    { title: "left", type: "range", min: 300, max: 900, value: 750 },
    { title: "right", type: "range", min: 300, max: 900, value: 750 },
    { title: "TLBR", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "TRBL", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "show arcs", type: "bool", value: false },
    { title: "distance", type: "range", min: 10, max: 1000, value: 50 },
    { title: "subtract", type: "bool", value: true }
];

module.exports = all;
```

closed 2019-09-08 by danmarshall

Neat-1, 2019-09-27:
Hi Dan 
Many thanks for your valued help I am getting fillets on outside using outline but not on inside, not too sure were to start with this one or if it is available?

Thank you in advance

danmarshall, 2019-09-27:
@Neat-1 can you share your drawing code or give an example (screenshot perhaps?)

Neat-1, 2019-09-27:
Hi Dan
The code above will produce this Pic
[image]

danmarshall, 2019-09-27:
Ok, so you may want to progress to the next level - using [chains](https://maker.js.org/docs/working-with-chains/#content). 
See https://maker.js.org/docs/working-with-chains/#Chain%20fillet

Neat-1, 2019-09-27:
Ok Thanks Dan I was hoping there was a way to make the inside offset follow the outsides and follow the fillets only because I have the ark intersection
But I can see the chains should do the job I will study the truss example
Thank you
Chris

Neat-1, 2019-09-28:
Hi Dan
Just curiosity when using the outline & true the 0 doesn't change?           (Use 0 for round joints}
Chris

danmarshall, 2019-10-06:
Hi Chris, yes 0 is the default.

Neat-1, 2019-10-06:
Hi Dan

 

I cant seem to get this working when (True) is set

 

 

 

Regards

Chris

 

 

 

 

 

Neatline fitted furniture        *      78 Church Street Radcliffe Manchester M26 2SY  

*         0161 723 1656                                                          

                                                            *        Neatline   

                                                            *l       mailto:c@neatline.co.uk <mailto:cgriff@neatline.co.uk> 

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Neat-1, 2019-10-06:
Hi Dan
When on outline true ( 0 ) does not seem to work?

danmarshall, 2019-10-06:
Not sure what you mean. What is the result you are getting, and what is the expected result?

Neat-1, 2019-10-06:
Hi Dan
Sorry should of explained better, the code below
    this.models = {
        shape, offset: makerjs.model.outline(shape, distance, 0, true)

when using the outline option (0) with  (true) the inside shape does not change between sharp corner, radius, and chamfer   hope this makes sense.

danmarshall, 2019-10-07:
So it helps to understand how outlining works. First, every path can be outlined with a "sausage" - a model that wraps around a path:
[image]
See https://maker.js.org/docs/advanced-drawing/#Expanding%20paths

The red is the shape of the exact distance surrounding the path.

This happens for every path in a model. So, here is a star:
[image]

Then all the sausages get unioned:
[image]

Notice how the outer outline has round "outer" joints, and sharp inner joints. The inside outline is the reverse. That's just the nature of it! So, if either of these shapes was all-rounded, it wouldn't be the exact distance surrounding the original star.

Neat-1, 2019-10-07:
Hi Dan

 

Thank you very much for going to the trouble to explain exactly how the outline works, I do understand now sorry for being an absolute pain 

 

 

 

 

 

Regards

Chris

 

 

 

 

 

Neatline fitted furniture        *      78 Church Street Radcliffe Manchester M26 2SY  

*         0161 723 1656                                                          

                                                            *        Neatline   

                                                            *l       mailto:c@neatline.co.uk <mailto:cgriff@neatline.co.uk> 

 

 <https://user-images.githubusercontent.com/11507384/66278472-2497aa80-e85e-11e9-8dbc-29fd303b9214.png> 

 <https://user-images.githubusercontent.com/11507384/66278496-72acae00-e85e-11e9-9c1e-ec6fd450a44e.png> 

 <https://user-images.githubusercontent.com/11507384/66278529-c0c1b180-e85e-11e9-9c9f-3693a99dafaf.png> 

https://www.avast.com/antivirus

Neat-1, 2019-10-12:
Hi Dan
Can you give me some direction on this code I have the first part working ok, I am trying to get the offset using your truss example, but I am struggling linking them together,I realize I am limited by my knowledge of js but I am still working on this. any help is so much appreciated Thank you ,Chris

var makerjs = require('makerjs');

function shape(bottom, left, right, diag1, diag2, showArcs) {
    this.units = makerjs.unitType.Millimeter;

    this.paths = {
        bottom: new makerjs.paths.Line([0, 0], [bottom, 0]),
        leftArc: new makerjs.paths.Arc([0, 0], left, 0, 180),
        rightArc: new makerjs.paths.Arc([bottom, 0], right, 0, 180),
        diagArc1: new makerjs.paths.Arc([bottom, 0], diag1, 135, 180),
        diagArc2: new makerjs.paths.Arc([0, 0], diag2, 0, 90)
       
       
    };
    var int1 = makerjs.path.intersection(this.paths.leftArc, this.paths.diagArc1);
    console.log(int1);
    this.paths.left = new makerjs.paths.Line([0, 0], int1.intersectionPoints[0]);

    var int2 = makerjs.path.intersection(this.paths.rightArc, this.paths.diagArc2);
    console.log(int2);
    this.paths.right = new makerjs.paths.Line([bottom, 0], int2.intersectionPoints[0]);

    this.paths.top = new makerjs.paths.Line(this.paths.left.end, this.paths.right.end);

    
    //Show Arcs
    if (!showArcs) {
        delete this.paths.leftArc;
        delete this.paths.rightArc;
        delete this.paths.diagArc1;
        delete this.paths.diagArc2;
    }
}

/*this.models.frame.origin = [22,22];
 
  
  var angled = this.models.frame.paths.ShapeLine1;

  var bracepoints = [
    [w/2, 22],
    //m.point.middle(angled, 1/1),
   // [w/2 , h],
    
   m.point.middle(angled, 1/2)
  ];

  this.models.brace = new m.models.ConnectTheDots(false, bracepoints);

var truss = new trussWireframe(1200, 800);
var expansion = m.model.expandPaths(truss, 22, 1);

m.model.originate(expansion);
m.model.simplify(expansion);

		

//find chains
var chains = m.model.findChains(expansion);

//start at 1 - ignore the longest chain which is the perimeter
for (var i = 1; i < chains.length; i++) {

  //save the fillets in the model tree 
  expansion.models['fillets' + i] = m.chain.fillet(chains[i], 3);
}

*/

shape.metaParameters = [
    { title: "bottom", type: "range", min: 300, max: 3000, value: 2000 },
    { title: "left", type: "range", min: 300, max: 900, value: 750 },
    { title: "right", type: "range", min: 300, max: 900, value: 750 },
    { title: "TLBR", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "TRBL", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "show arcs", type: "bool", value: false },
    { title: "distance", type: "range", min: 10, max: 1000, value: 50 },
    { title: "subtract", type: "bool", value: true }
];

module.exports = shape;

danmarshall, 2019-10-13:
Hi @Neat-1 , here's how I did it:
```js
var makerjs = require('makerjs');

function shape(bottom, left, right, diag1, diag2) {
    this.units = makerjs.unitType.Millimeter;
	var leftArc = new makerjs.paths.Arc([0, 0], left, 0, 180);
	var rightArc = new makerjs.paths.Arc([bottom, 0], right, 0, 180);
	var diagArc1 = new makerjs.paths.Arc([bottom, 0], diag1, 135, 180);
    var diagArc2 = new makerjs.paths.Arc([0, 0], diag2, 0, 90);
    this.paths = {
        bottom: new makerjs.paths.Line([0, 0], [bottom, 0]),
    };
    var int1 = makerjs.path.intersection(leftArc, diagArc1);
    this.paths.left = new makerjs.paths.Line([0, 0], int1.intersectionPoints[0]);
    var int2 = makerjs.path.intersection(rightArc, diagArc2);
    this.paths.right = new makerjs.paths.Line([bottom, 0], int2.intersectionPoints[0]);
    this.paths.top = new makerjs.paths.Line(this.paths.left.end, this.paths.right.end);
}

function offset(shape, distance) {
	this.models = {
		shape, offset: makerjs.model.outline(shape, distance, 0, true)
	}
	var midTop = makerjs.point.middle(shape.paths.top);
	var midBottom = makerjs.point.middle(shape.paths.bottom);
	this.models.slot = new makerjs.models.Slot(midTop, midBottom, distance / 2);
	makerjs.model.combineSubtraction(this.models.offset, this.models.slot)
}

function all(bottom, left, right, tlbr, trbl, distance, outerFilletDistance, innerFilletDistance) {
    var s = new shape(bottom, left, right, tlbr, trbl);
    var o = new offset(s, distance);

	//new code below
  	var chains = makerjs.model.findChains(o);
  
  	//the first chain is the longest.
  	var outerFillets = makerjs.chain.fillet(chains[0], outerFilletDistance);

  	//we know there are 2 inner chains
	var innerFillets1 = makerjs.chain.fillet(chains[1], innerFilletDistance);
	var innerFillets2 = makerjs.chain.fillet(chains[2], innerFilletDistance);

  
	this.models = {o, outerFillets, innerFillets1, innerFillets2};
}

all.metaParameters = [
    { title: "bottom", type: "range", min: 300, max: 3000, value: 2000 },
    { title: "left", type: "range", min: 300, max: 900, value: 750 },
    { title: "right", type: "range", min: 300, max: 900, value: 750 },
    { title: "TLBR", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "TRBL", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "distance", type: "range", min: 10, max: 1000, value: 50 },
    { title: "outer fillet", type: "range", min: 0, max: 50, value: 20 },
    { title: "inner fillet", type: "range", min: 0, max: 50, value: 20 }
];

module.exports = all;
```

[image]

Neat-1, 2019-10-15:
Hi Dan
Perfect many thanks for this, it's the inner squares I use to cut the panels from the Dxf is there a way to hide the outer line, I have tried (delete this) but I can only remove two of them? 
Kind Regards

var makerjs = require('makerjs');

function shape(bottom, left, right, diag1, diag2) {
    this.units = makerjs.unitType.Millimeter;
	var leftArc = new makerjs.paths.Arc([0, 0], left, 0, 180);
	var rightArc = new makerjs.paths.Arc([bottom, 0], right, 0, 180);
	var diagArc1 = new makerjs.paths.Arc([bottom, 0], diag1, 135, 180);
    var diagArc2 = new makerjs.paths.Arc([0, 0], diag2, 0, 90);
    this.paths = {
        bottom: new makerjs.paths.Line([0, 0], [bottom, 0])
    };

    var int1 = makerjs.path.intersection(leftArc, diagArc1);
    this.paths.left = new makerjs.paths.Line([0, 0], int1.intersectionPoints[0]);
    var int2 = makerjs.path.intersection(rightArc, diagArc2);
    this.paths.right = new makerjs.paths.Line([bottom, 0], int2.intersectionPoints[0]);
    this.paths.top = new makerjs.paths.Line(this.paths.left.end, this.paths.right.end);
}

function offset(shape, distance) {
	this.models = {
		shape, offset: makerjs.model.outline(shape, distance, 0, true)
	}
	var midTop = makerjs.point.middle(shape.paths.top);
	var midBottom = makerjs.point.middle(shape.paths.bottom);
	this.models.slot = new makerjs.models.Slot(midTop, midBottom, distance / 2);
	makerjs.model.combineSubtraction(this.models.offset, this.models.slot);
    
   delete shape.paths.top;
   delete shape.paths.left;
   //delete shape.paths.bottom;
   //delete shape.paths.right;
  
}

 
function all(bottom, left, right, tlbr, trbl, distance, outerFilletDistance, innerFilletDistance) {
    var s = new shape(bottom, left, right, tlbr, trbl);
    var o = new offset(s, distance);

	//new code below
  	var chains = makerjs.model.findChains(o);
  
  	//the first chain is the longest.
  	var outerFillets = makerjs.chain.fillet(chains[0], outerFilletDistance);

  	//we know there are 2 inner chains
	var innerFillets1 = makerjs.chain.fillet(chains[1], innerFilletDistance);
	var innerFillets2 = makerjs.chain.fillet(chains[2], innerFilletDistance);

     
	this.models = {o, outerFillets, innerFillets1, innerFillets2};
}

all.metaParameters = [
    { title: "bottom", type: "range", min: 300, max: 3000, value: 2000 },
    { title: "left", type: "range", min: 300, max: 900, value: 750 },
    { title: "right", type: "range", min: 300, max: 900, value: 750 },
    { title: "TLBR", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "TRBL", type: "range", min: 700, max: 3000, value: 1800 },
    { title: "distance", type: "range", min: 10, max: 1000, value: 22},
    { title: "outer fillet", type: "range", min: 0, max: 50, value: 0},
    { title: "inner fillet", type: "range", min: 0, max: 50, value: 5 }
];

module.exports = all;

[image]

danmarshall, 2019-10-15:
Yes, using https://maker.js.org/docs/working-with-chains/#Chain%20to%20new%20model . You can pass the largest chain as the first parameter, and true as the second parameter which means “detach” the chain from its current model(s).

Neat-1, 2019-10-15:
Brilliant thank you