microsoft/maker.js #438: function part of code not showing in svg

Neat-1, 2019-09-07:
Hi
The function in this code is not showing on svg I am sure its something I have missed but I can't
seem to find the solution.
Many thanks in advance
Chris

var m = require('makerjs');
function cg(w, h, w1, h1,width,height) {
    this.models = {
        shape1: new m.models.ConnectTheDots(true, [[w, h], [1600, 0], [0, 0], [w1, h1]])
    };    
   this.models = {
     example: new makerjs.models.Rectangle(width, height)
    };
}   
  cg.metaParameters = [
    { title: "Top width", type: "range", min: 1, max: 2100, value: 1600 },
    { title: "Right Height", type: "range", min: 1, max: 2100, value: 1600 },
    { title: "Left Top Angle", type: "range", min: -200, max: 2100, value: 0 },
    { title: "Left Height", type: "range", min: 0, max: 2100, value: 1600 },
    { title: "width", type: "range", min: 100, max: 300, value: 150 },
    { title: "height", type: "range", min: 100, max: 222, value: 150 }
];
module.exports = cg;

danmarshall, 2019-09-08:
Seems to work for me:
[image]

Neat-1, 2019-09-08:
Hi Dan
The first 4 parameters is the control of the function, and the bottom 2 parameters is for the rectangle
I can not get the two shapes on svg together? hope that makes sense

danmarshall, 2019-09-09:
Oh, right. It's because you are reassigning `this.models`. Instead of :
```
this.models = {
  shape1: new m.models.ConnectTheDots(true, [[w, h], [1600, 0], [0, 0], [w1, h1]])
};
this.models = {
  example: new makerjs.models.Rectangle(width, height)
};
```
try:
```
this.models = {
  shape1: new m.models.ConnectTheDots(true, [[w, h], [1600, 0], [0, 0], [w1, h1]]),
  example: new makerjs.models.Rectangle(width, height)
};
```

closed 2019-09-09 by danmarshall