microsoft/maker.js #440: Feature request:

Neat-1, 2019-09-15:
Just like to say what a great  resource
Would be great to have the parameters boxes to sit along side each other , you could keep  x - y
co-ordinates  together and perhaps not include the slider part.
I don't know if this is something you can do at the moment ?

danmarshall, 2019-09-16:
I've thought about this before, specifically for x,y coords. So, I think it could be a new primitive: a Point input. Then, instead of a slider, we could have a "point picker" which is just letting the user mouse over the canvas and picking a point.

Neat-1, 2019-09-17:
Hi Dan 
That's quite interesting it would save a lot of space also (maybe if possible two x y alongside each other) would it be a just line or could it be an ark from last point.
I have done a simple adjustment in css to allow the  parameters to scroll to help with space pic below
[image]