microsoft/maker.js #442: Show an example opentype.RenderOptions object

doox911, 2019-09-26:
Hey. I don’t understand which object to transfer as the last argument to 

`new Text (font: opentype.Font, text: string, fontSize: number, combine ?: boolean, centerCharacterOrigin ?: boolean, bezierAccuracy ?: number, opentypeOptions ?: opentype.RenderOptions): Text, `

because I have `opentype.RenderOptions === undefined`?

danmarshall, 2019-11-21:
Hi @doox911 - from your example, it seems that you might want [captions](https://maker.js.org/docs/intermediate-drawing/#Captions) instead of [fonts](https://maker.js.org/docs/advanced-drawing/#Fonts%20and%20text)

>Captions are fragments of text that can be positioned anywhere in your model, useful for adding documentation within your drawing. Captions are unlike the Text model, which is a line drawing of glyphs in a given font.

utkarshsaraogi, 2019-11-27:
Can you exactly tell what are those options? I also looking for this, since I need to generate text that has fill property not just outline.

danmarshall, 2019-11-27:
@utkarshsaraogi The [TextModel code invokes](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/models/Text.ts#L62) an opentype.js font's [forEachGlyph method](https://github.com/opentypejs/opentype.js/blob/master/src/font.js#L294) which takes the options object. 

This object is loosely defined by opentype.js in [their Readme for the font object](https://github.com/opentypejs/opentype.js#the-font-object), and further defined in [their JSDoc](https://github.com/opentypejs/opentype.js/blob/master/src/font.js#L262) as:

```
/**
 * @typedef GlyphRenderOptions
 * @type Object
 * @property {string} [script] - script used to determine which features to apply. By default, 'DFLT' or 'latn' is used.
 *                               See https://www.microsoft.com/typography/otspec/scripttags.htm
 * @property {string} [language='dflt'] - language system used to determine which features to apply.
 *                                        See https://www.microsoft.com/typography/developers/opentype/languagetags.aspx
 * @property {boolean} [kerning=true] - whether to include kerning values
 * @property {object} [features] - OpenType Layout feature tags. Used to enable or disable the features of the given script/language system.
 *                                 See https://www.microsoft.com/typography/otspec/featuretags.htm
 */
``` 
Looking at their [forEachGlyph method code](https://github.com/opentypejs/opentype.js/blob/master/src/font.js#L294) we can see that it also has some other undocumented options like [letterSpacing](https://github.com/opentypejs/opentype.js/blob/master/src/font.js#L322) and [tracking](https://github.com/opentypejs/opentype.js/blob/master/src/font.js#L324).

But remember, Maker.js does not have a "fill" concept internally - it is strictly for line drawing without fill, until you export your line drawing to a fill-enabled format such as SVG. So these opentype.js options might not help you achieve that.

Perhaps you can let me know what you are trying to achieve, it may be that you want captions instead of letterform outlines (fonts) if you want filled text (see my previous comment above). It also depends on your output format. Are you using SVG?

utkarshsaraogi, 2019-11-27:
I am creating single line diagram(SLD) using maker.js. There are various components and connection wires having text labels. I am exporting this model as SVG to extract all the paths data, to render it in my app, so that user can interact with it. Like zoom in/out and pan. Also can provide input to generate the SLD accordingly.

By default loading font file from opentype.js and then using it to create text model generates a text that has outline boundary, which is not very crisp/clear when zoomed out a bit. 

I am looking for a way so that I can generate text which contains only single path(line), not double outlines or if they can be filled with some color.

danmarshall, 2019-11-29:
@utkarshsaraogi it does sound that captions would be better suited to your scenario. Let me know we can make the size work for you.