microsoft/maker.js #445: Maker Locally

Neat-1, 2019-10-25:
Hi All
Can anybody point me in the right direction to install maker locally, what files will be required to run it.
Thank you

danmarshall, 2019-11-21:
Hi @Neat-1, sorry for the late reply. Here's how you can run the Playground locally:
```
git clone https://github.com/microsoft/maker.js.git
cd maker.js
npm install
npm run build
npm start
```
then point your browser to http://localhost:8020/docs/playground/index.html

danmarshall, 2019-11-21:
Hi @Neat-1 , I just created a new repo which makes a very nice environment for local development:
https://github.com/danmarshall/makerjs-simple-page

Please give it a try and let me know what you think.

Neat-1, 2019-11-24:
Hi Dan

Many thanks for both types of information to use maker locally, I have not managed to open simple-page yet showing this error still trying

Thank you

Chris

 

https://github.com/danmarshall/makerjs-simple-page

Neat-1, 2019-11-24:
Hi Dan

 

Just the smile face showing on local view this is code from c prompt

Chris

 

C:\makerjs-simple-page-master>npm  install

npm WARN makerjs-simple-page@1.0.0 No repository field.

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules\fsevents):

npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

 

audited 8381 packages in 10.869s

found 0 vulnerabilities

 

 

C:\makerjs-simple-page-master>npm start

 

> makerjs-simple-page@1.0.0 start C:\makerjs-simple-page-master

> parcel index.html --open

 

Server running at http://localhost:1234

√  Built in 1.05s.

 

https://github.com/danmarshall/makerjs-simple-page

danmarshall, 2019-11-25:
Hi @neat-1 maybe you can clarify your goal. The makerjs-simple-page is a lightweight environment for you to write code using your favorite code editor locally. I’d say running the Playground locally is not any more advantageous than the online version (unless you need to be disconnected from the internet).

Neat-1, 2019-11-25:
Hi Dan

Thank you for your email, my goal is to place my code in editor from saved file, adjust parameters, run maker, then save As customers name 

I can do this online but must copy & paste into editor then download dxf and change name.  

Unless there is something, I am missing

 

Kind Regards 

Chris

danmarshall, 2019-11-26:
Hi Chris, in your case I think you can automate to the point where you don't even need a web page. You can just run your code as a Node.js script, take parameters & customer name, and save as a DXF file all from from the command line.

closed 2021-07-17 by danmarshall