microsoft/maker.js #446: Error building playground

sent-hil, 2019-11-08:
I'm very new to typescript and nodejs, so this might not be an error, but just warnigns.

After doing `npm i`, I ran `tsc` and go this:

```
~/p/c/m/p/playground:master$ tsc --version
Version 3.6.4
~/p/c/m/p/playground:master$ tsc
src/pointer.ts:93:94 - error TS2352: Conversion of type 'PointerEvent' to type 'IPointerEvent' may be a mistake because neither type sufficiently overlaps with the other. If this was intentional, convert the expression to 'unknown' first.
  Types of property 'srcElement' are incompatible.
    Type 'EventTarget' is missing the following properties from type 'Element': assignedSlot, attributes, classList, className, and 120 more.

93             view.addEventListener('pointerdown', (e: PointerEvent) => { this.viewPointerDown(e as IPointerEvent); });
                                                                                                ~~~~~~~~~~~~~~~~~~

src/pointer.ts:94:94 - error TS2352: Conversion of type 'PointerEvent' to type 'IPointerEvent' may be a mistake because neither type sufficiently overlaps with the other. If this was intentional, convert the expression to 'unknown' first.

94             view.addEventListener('pointermove', (e: PointerEvent) => { this.viewPointerMove(e as IPointerEvent); });
                                                                                                ~~~~~~~~~~~~~~~~~~

src/pointer.ts:95:90 - error TS2352: Conversion of type 'PointerEvent' to type 'IPointerEvent' may be a mistake because neither type sufficiently overlaps with the other. If this was intentional, convert the expression to 'unknown' first.

95             view.addEventListener('pointerup', (e: PointerEvent) => { this.viewPointerUp(e as IPointerEvent); });
                                                                                            ~~~~~~~~~~~~~~~~~~

src/pointer.ts:379:55 - error TS2345: Argument of type 'WheelEvent' is not assignable to parameter of type 'IMouseEvent'.
  Types of property 'srcElement' are incompatible.
    Type 'EventTarget' is missing the following properties from type 'Element': assignedSlot, attributes, classList, className, and 120 more.

379             var pointRelative = this.getPointRelative(e);
                                                          ~

src/pointer.ts:391:17 - error TS2322: Type 'EventTarget' is not assignable to type 'Element'.

391                 srcElement: e.srcElement
                    ~~~~~~~~~~

  src/pointer.ts:25:9
    25         srcElement: Element;
               ~~~~~~~~~~
    The expected type comes from property 'srcElement' which is declared here on type 'IPointer'

src/pointer.ts:393:27 - error TS2339: Property 'wheelDelta' does not exist on type 'WheelEvent'.

393             var sign = (e.wheelDelta || e['deltaY']) > 0 ? 1 : -1;
                              ~~~~~~~~~~

src/require-iframe.ts:202:5 - error TS2322: Type '(id: string) => any' is not assignable to type 'NodeRequireFunction & NodeRequire'.
  Type '(id: string) => any' is missing the following properties from type 'NodeRequire': resolve, cache, extensions, main

202     window.require = function (id: string) {
        ~~~~~~~~~~~~~~

Found 7 errors.
```

Even thought compiler outputter "errors", it did change few files in output dir:

```
~/p/c/maker.js:master$ gs
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   docs/playground/js/format-options.js
        modified:   docs/playground/js/format-options.js.map
        modified:   docs/playground/js/playground.js.map
        modified:   docs/playground/js/pointer.js.map
        modified:   docs/playground/js/require-iframe.js.map
```

Is this expected or perhaps I'm doing something wrong?

danmarshall, 2019-11-08:
Hi @sent-hil , can you try to `npm run build` from the root?

sent-hil, 2019-11-09:
Here you go:

```
~/p/c/maker.js:master$ npm run build
lerna ERR! npm run build exited 2 in 'fonts'
lerna ERR! npm run build stdout:

> fonts@1.0.0 build /Users/senthilarivudainambi/play/cnc/maker.js/packages/fonts
> tsc -p .

../../node_modules/@types/opentype.js/index.d.ts(34,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'.
../../node_modules/@types/opentype.js/index.d.ts(42,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'.
../../node_modules/@types/opentype.js/index.d.ts(50,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'.
../../node_modules/@types/opentype.js/index.d.ts(212,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'.
../../node_modules/@types/opentype.js/index.d.ts(219,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'.
../../node_modules/@types/opentype.js/index.d.ts(226,14): error TS2304: Cannot find name 'CanvasRenderingContext2D'.
../../node_modules/@types/opentype.js/index.d.ts(343,15): error TS2304: Cannot find name 'CanvasRenderingContext2D'.
../../node_modules/@types/opentype.js/index.d.ts(350,42): error TS2304: Cannot find name 'SVGPathElement'.

lerna ERR! npm run build stderr:
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! fonts@1.0.0 build: `tsc -p .`
npm ERR! Exit status 2
npm ERR!
npm ERR! Failed at the fonts@1.0.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/senthilarivudainambi/.npm/_logs/2019-11-09T07_02_40_720Z-debug.log

lerna ERR! npm run build exited 2 in 'fonts'
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! makerjs-dev@1.0.0 build: `lerna run build`
npm ERR! Exit status 2
npm ERR!
npm ERR! Failed at the makerjs-dev@1.0.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/senthilarivudainambi/.npm/_logs/2019-11-09T07_02_40_850Z-debug.log
```

Run on latest master.

sent-hil, 2019-11-09:
Looks like that succeeded:

```
~/p/c/maker.js:master$ npm run build

> makerjs-dev@1.0.0 build /Users/senthilarivudainambi/play/cnc/maker.js
> lerna run build

lerna notice cli v3.18.4
lerna info Executing command in 3 packages: "npm run build"
lerna info run Ran npm script 'build' in 'fonts' in 3.9s:

> fonts@1.0.0 build /Users/senthilarivudainambi/play/cnc/maker.js/packages/fonts
> tsc -p .

lerna info run Ran npm script 'build' in 'makerjs' in 13.1s:

> makerjs@0.17.0 prebuild /Users/senthilarivudainambi/play/cnc/maker.js/packages/maker.js
> npm run gen-cascades

> makerjs@0.17.0 gen-cascades /Users/senthilarivudainambi/play/cnc/maker.js/packages/maker.js
> npm run project && node ./target/cascadable.js > ./src/core/cascades.ts

> makerjs@0.17.0 project /Users/senthilarivudainambi/play/cnc/maker.js/packages/maker.js
> typedoc ./src --json ./dist/project.json

Using TypeScript 2.7.2 from /Users/senthilarivudainambi/play/cnc/maker.js/node_modules/typedoc/node_modules/typescript/lib

JSON written to /Users/senthilarivudainambi/play/cnc/maker.js/packages/maker.js/dist/project.json

> makerjs@0.17.0 build /Users/senthilarivudainambi/play/cnc/maker.js/packages/maker.js
> tsc -p target && node ./target/build.js

lerna info run Ran npm script 'build' in 'playground' in 6.3s:

> playground@1.0.0 build /Users/senthilarivudainambi/play/cnc/maker.js/packages/playground
> tsc -p . && tsc -p src/worker

lerna success run Ran npm script 'build' in 3 packages in 19.5s:
lerna success - fonts
lerna success - makerjs
lerna success - playground
```

closed 2019-11-21 by danmarshall