microsoft/maker.js #447: Convert to canvas polygon to DXF Please help

azeemgujjar, 2019-11-18:
I have polygon on canvas, i want to export it as DXF, no fiddle or any working example i found online please help me.

danmarshall, 2019-11-18:
Hello @azeemgujjar , can you please provide more details? A code sample would be nice.
I'm not clear on what you mean by "canvas" - if you mean a rasterized "bitmap" then Maker.js might not  help you, as it does not import from non-vector based sources.

azeemgujjar, 2019-11-19:
> Hello @azeemgujjar , can you please provide more details? A code sample would be nice.
> I'm not clear on what you mean by "canvas" - if you mean a rasterized "bitmap" then Maker.js might not help you, as it does not import from non-vector based sources.

Hey! thanks Dan for your response, actually i am creating polygon shape using html5 canvas in js.
but i want it to export that shape in DXF format, no other library i found which support export html5 canvas drawing to DXF format.

For example

`<canvas id="myCanvas" width="700" height="410">
  <p>Some default content can appear here.</p>
</canvas>
<p>Triangles!</p>`

`function draw() {
  // canvas with id="myCanvas"
  var canvas = document.getElementById('myCanvas');
  if (canvas.getContext) {
    var ctx = canvas.getContext('2d');
    ctx.beginPath(); // note usage below 

    // triangle 1, at left
    ctx.fillStyle = "#F9A520";
    ctx.moveTo(0, 0); // start at top left corner of canvas
    ctx.lineTo(200, 0); // go 200px to right (x), straight line from 0 to 0
    ctx.lineTo(100, 200); // go to horizontal 100 (x) and vertical 200 (y)
    ctx.fill(); // connect and fill

    // triangle 2, top center
    ctx.moveTo(300, 0); // pick up "pen," reposition at 300 (horiz), 0 (vert)
    ctx.lineTo(300, 200); // draw straight down (from 300,0) to 200px
    ctx.lineTo(500, 100); // draw up toward right (100 half of 200)
    ctx.fill(); // connect and fill

    // triangle 3, bottom center
    ctx.beginPath(); // note: w/o this, color does not work as expected 
    ctx.fillStyle = "#F00";
    ctx.moveTo(300, 200); // pick up "pen," reposition at 300 (horiz), 200 (vert)
    ctx.lineTo(300, 400); // draw straight down by 200px (200 + 200)
    ctx.lineTo(100, 300); // draw up toward left (100 less than 300, so left)
    ctx.fill(); // connect and fill

    // triangle 4, at right, top
    ctx.beginPath();
    ctx.strokeStyle = "#00F";
    ctx.moveTo(600, 0); // pick up "pen," reposition at 500 (horiz), 0 (vert)
    ctx.lineTo(500, 200); // draw straight down by 200px (200 + 200)
    ctx.lineTo(700, 200); // draw up toward left (100 less than 300, so left)
    ctx.closePath(); // connect end to start
    ctx.stroke(); // outline the shape that's been described

    // triangle 5, at right, bottom
    ctx.beginPath();
    ctx.strokeStyle = "#096";
    ctx.moveTo(500, 210); // pick up "pen," reposition 
    ctx.lineTo(700, 210); // draw straight across to right
    ctx.lineTo(600, 410); // draw down toward left
    ctx.closePath(); // connect end to start
    ctx.stroke(); // outline the shape that's been described

  }
}

draw();`

danmarshall, 2019-11-19:
Hi @azeemgujjar , thanks for the code sample. Is there any reason you can't use SVG instead of canvas?

For canvas, there are a couple of ways I can imagine to accomplish this. First, convert your drawing commands to Maker.js objects. Then you can either:
* Export as SVG, then convert SVG to canvas using another library, such as Fabric https://www.npmjs.com/package/fabric
* Traverse your Maker.js model tree and generate canvas commands. https://maker.js.org/docs/model-trees/#content

 There's some caveats you will need to be aware of:
1. Maker.js does not have a concept of "fill" since it is strictly for line drawing. The DXF export function also does not export any fill. I'm not certain if the DXF format itself has this concept or not.
1. If you are exporting as SVG, you can specify fill colors by layer. https://maker.js.org/docs/exporting/#SVG If you are generating canvas commands, you can add the "fill" color value as an extra property on your objects, which you can then pick up when you convert to canvas commands.
1. Maker.js Y coordinates are opposite of canvas & SVG: https://maker.js.org/docs/basic-drawing/#Points

azeemgujjar, 2019-11-20:
HI @danmarshall , thank you for your response. Yes i only want to export line drawing without "Fill".
but problem is first i have to convert all my script to fabric.js canvas which is annoying, plus maker.js doesn't support all type of SVG's only specific type of SVG formats supported (want i understand after watching your documentation and features).
Anyways thank you so much

danmarshall, 2019-11-21:
You may want to look at https://github.com/d3/d3-path which can serialize CanvasPathMethods calls to SVG path data.

closed 2021-07-17 by danmarshall

tranthaihoang, 2023-10-30:
hi @azeemgujjar 
Were you able to solve your problem? I also have a similar situation, I want to convert the lines in the canvas into a DXF file.