microsoft/maker.js #448: Booleans on Bezier arcs with large radius

hypercross, 2019-12-06:
I was dealing with path booleans with maker.js recently, when I encountered a problem where a boolean operation results in fragmented segments that end up being discarded, resulting in entire holes or contours lost.

I did some digging, and it seems that the Text objects I was dealing with are in fact Bezier Curves approximated with arcs. When the bezier curve goes flat, the arc's radius becomes huge and the angle becomes very small. That seemed to cause issues when the arc is intersected with lines.

I converted the arcs in question into straight lines and that seemed to solve the problem. But I guess a better solution would have to be implemented by the devs.

danmarshall, 2019-12-06:
Hello, thanks for reporting this issue. I've also noticed similar behavior and am trying to fix it. Can you provide any specific code example to repro the issue? I'd like to make a unit test for it.

hypercross, 2019-12-07:
I made this in the maker.js playground. 

```javascript
const makerjs = require('makerjs');
const {Oval, Square} = makerjs.models;
const {combineUnion, combineSubtraction} = makerjs.model;

// const CIRCLE_DIAMETER = 4000;
// const SQUARE_SIDE   = 100;

function booleanTest(CIRCLE_DIAMETER, SQUARE_SIDE){
  
  let oval = new Oval(CIRCLE_DIAMETER, CIRCLE_DIAMETER);
  
  // making inner ring
  for(let i = 1; i < 4; i ++){
    const innerOval = new Oval(CIRCLE_DIAMETER - i * 10, CIRCLE_DIAMETER - i * 10);
    innerOval.origin = [i * 5, i * 5];
    const op = i % 2 ? combineSubtraction : combineUnion;
    oval = op(oval, innerOval);
  }

  // square on the edge
  const square = new Square(SQUARE_SIDE);
  square.origin = [CIRCLE_DIAMETER - SQUARE_SIDE * .5, CIRCLE_DIAMETER * .5 - SQUARE_SIDE * .5 ];

  this.models = {
      result: combineSubtraction(oval, square)
  };

  this.origin = [CIRCLE_DIAMETER * -1, CIRCLE_DIAMETER * -.5];

  this.notes = 'The shape consists of 2 concentric rings, being cut on the edge with a square.';
  this.notes += 'We should see 2 rings with opening after the cut.';
}

booleanTest.metaParameters = [
  { title: "circle diameter", type: "range", min: 500, max: 5000, value: 1700 },
  { title: "square side", type: "range", min: 50, max: 300, value: 50 },
];

module.exports = booleanTest;
```