microsoft/maker.js #450: Dynamically modifying the range limits of a metaParameter

yertto, 2020-01-27:
Its currently possible to initialize the `max` limit of a `range` `metaParameter` with a number:
 * https://maker.js.org/docs/api/interfaces/makerjs.imetaparameter.html#max

But, would it be possible to dynamically modify the `max` based on another value by maybe passing in a function instead?

eg. 
 * https://maker.js.org/playground/?script=https://gist.githubusercontent.com/yertto/bf3fde0e78ea3b9b36000d7a55b14677/raw/c9914d84374ceb95f19083b02c31d5ac94d628fa/nut.js
 * https://gist.github.com/yertto/bf3fde0e78ea3b9b36000d7a55b14677

This is a made up example of a nut (🔩), where the aim is to limit the maximum hole size to always be smaller than the outer hexagon.

danmarshall, 2020-01-27:
Interesting issue. Is the context in terms of the UI (for example the Playground) or to guard against invalid parameters?

yertto, 2020-01-27:
Thanks @danmarshall,

Yes my context is in terms of just playing with `metaParameters` in the [Playground](https://maker.js.org/playground).

I guess my end users are one layer above the [Playground](https://maker.js.org/playground) (a "Meta Playground"?), as I'm not really interested in letting them edit code.

BTW, I saw https://github.com/microsoft/maker.js/issues/380#issuecomment-474065503 and had a look at [svg-blueprint](https://onlfait.github.io/svg-blueprint/demo/) - which says its customizable - but I didn't see a way of adding the "customize" UI to it that [Playground](https://maker.js.org/playground) has for the `metaParameters`.

However I am finding [Playground](https://maker.js.org/playground) insanely useful, even if the code editor fills half the screen, so I'm not too fussed by that.

danmarshall, 2020-01-27:
I also would like to have a no-code Playground. The Playground should be broken up into components so that you can compose your own experience. I think there's already an issue for this.

As for the dynamic range issue, sometimes you can take the "relative" approach. In your case, hole size would be a range from 0% to 100% of the outer hexagon. You might want to build your own UI for your app.

yertto, 2020-01-27:
Yes, I had a similar thought for the "relative" approach where I take relative "steps" down from the outer shape.

FYI - here's my actual use case:
 *  https://maker.js.org/playground/?script=https://raw.githubusercontent.com/yertto/champagne/v2.0/champagne.js

It _calculates_ the hole sizes (based on the "steps" - throwing away any calculated hole size < 0) and displays the results in the "notes" section.

However I wanted to explore this dynamic range idea to try making the existing UI a bit more friendly without having to go build my own.
ie. by letting the user directly adjust the hole sizes instead of making them figure out how my crazy "steps" work.

Also I noticed some of the demos in the playground seem to go a bit whacky with certain combinations of values.
(eg. the [makerjs-spokes-straight]( https://maker.js.org/playground/?script=makerjs-spokes-straight) when [outerRadis < innerRadius](https://maker.js.org/playground/?script=makerjs-spokes-straight#params=[10,30,10,5,67,0,0,true]))
So I thought perhaps this feature might be useful elsewhere.

BTW - while I've got your expert attention 😄- any tips or suggestions on how to make my code any better?
 * https://github.com/yertto/champagne/issues

danmarshall, 2020-01-27:
Oh do you mean the linting? Seems like the online linter doesn't have the latest language features. I think your code is ok 👌 but I should see if there is an update for the linter.