microsoft/maker.js #452: How to export SVG with right aligned text ?

amatyat, 2020-02-10:
How to export SVG with right-aligned text (Arabic)?
Below code generate left-aligned text :

var makerjs = require('makerjs')
var textModel;

opentype.load('fonts/Roboto-Black.ttf', function(err, font) {
    if (err)
    {
                alert('Could not load font: ' + err);
    }
    else
    {
		var text = 'Hello World! This box should start word-wrapping!';
                textModel = new makerjs.models.Text(font, text, 45, true, true);
		var svg = makerjs.exporter.toSVG(textModel,{stroke: '#000', fill: '#000'});
		document.write(svg);
    }

danmarshall, 2020-02-10:
Hello, just to clarify do you mean right-to-left text?

amatyat, 2020-02-10:
Nup, just right aligned text. Like below : 

[image]

danmarshall, 2020-02-10:
First, see [this issue comment](https://github.com/microsoft/maker.js/issues/349#issuecomment-410027287) about using the opentype-layout library for opentype.js

Then, check out [the readme for computeLayout](https://github.com/Jam3/opentype-layout/blob/master/README.md) which has an align option.

For right-to-left, this should be [supported by opentype.js](https://github.com/opentypejs/opentype.js/pull/361). It’s possible that [our copy of opentype.js](https://github.com/microsoft/maker.js/blob/master/docs/external/opentype/opentype.js) for the Playground may need to be updated, please let me know or send a PR 😁

amatyat, 2020-02-18:
For right-to-left, could you please share an example of how to use it?