microsoft/maker.js #453: Support for dotted and dashed line/path

neel-radica, 2020-02-17:
I believe another options for type path should be included to support exporting dashed/dotted lines as shown. Checked the API and also the code, but I don't think the options is introduced yet. In DXF, the group code is 49 [Group code 49](http://help.autodesk.com/view/ACD/2015/ENU/?guid=GUID-F57A316C-94A2-416C-8280-191E34B182AC)

[image]

danmarshall, 2020-02-17:
Hi @neel-radica, the current approach to styles is to put them on their own layer. Will a dashed layer be suitable for your scenario? Or is it important that these dashed lines are one the same layer as other entities?

neel-radica, 2020-02-17:
Hi @danmarshall assuming that I use layer, it seems like there is only 2 options available, color & fontSize. I'm trying to add a new options to support dash lines but facing some difficulties. Maybe will submit a pull request if possible.

danmarshall, 2020-02-17:
Yes, currently there is no dashed line. My question is would the above proposal work for you?

danmarshall, 2020-02-17:
I suggest that we add it to layerOptions. If we do this, it means that dashed lines will be a separate layer. Is this solution ok?

neel-radica, 2020-02-18:
Hi @danmarshall sorry for the late reply. Absolutely, that is just what we need. However, I'm curious about how the implementation will be. I have changed a bit of the source code to be able to accept a layerOptions for dashes, however I end up doing a lot of //@ts-ignore cause dxf-parse doesnt have interface to support dashes yet. Probably, it's better to leave it at your hands. If needed, I could submit a pull request just for your reference, just say the word. Thanks a lot

ascweb, 2020-12-20:
> I suggest that we add it to layerOptions. If we do this, it means that dashed lines will be a separate layer. Is this solution ok?

Hi @danmarshall do you have any update about line thickness, opacity, and dashed style? Thanks a lot!

danmarshall, 2021-01-11:
@ascweb - for DXF, we'll need to add an additional LineType - currently there is only a style named [CONTINUOUS](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/dxf.ts#L179)

For PDF, we'll need to implement the `.dash` [method in PDFKit](http://pdfkit.org/docs/vector.html)

ascweb, 2021-01-16:
> Hi @danmarshall sorry for the late reply. Absolutely, that is just what we need. However, I'm curious about how the implementation will be. I have changed a bit of the source code to be able to [...]

@neel-radica did you get a good solution for dot/dashed lines?

danmarshall, 2026-01-08:
Support for DXF dashed lines added in 0.18.2