microsoft/maker.js #454: Trying to add Maker.JS to Wordpres...

ChrisD2012, 2020-02-17:
Hi this is not so much a problem with Maker.JS, it looks amazing and from the playgrounds and instructions, I think I am almost there.

Basically I am trying to use Make.JS on a simple template page on wordpress.
I have uploaded the following to my site folders.
browser.maker.js
bezier.js

From there I am trying to take the simple demo for https://maker.js.org/playground/
To display this on the template page.
I am trying to place it in the body of a HTML using Java Script, but every time it just comes up blank with no errors in the console.

Any help would be welcomed on this.

danmarshall, 2020-02-17:
Hi @ChrisD2012 welcome to the community. Please take a look at #380 and #398 if those are relevant to your question. If you are looking for the blueprint UI, many others are too.
To give a birds eye view, Maker.js is somewhat low level and does not make assumptions about the type of environment it runs in, so there’s really no UI built in. Which is cool because that makes it as lightweight as possible. Also it’s neat that it runs “headless” and can integrate into a backend as well as frontend.

danmarshall, 2021-07-17:
You may want to look at the way it's done in Glitch for some guidance: https://github.com/microsoft/maker.js/discussions/491#discussioncomment-407504

closed 2021-07-17 by danmarshall