microsoft/maker.js #455: "TypeError: Cannot read property 'low' of undefined" - When running combine with pointMatchingDistance

Bersaelor, 2020-02-18:
I've been making great progress with maker.js but hit a problem now.

### Makerjs version used: `makerjs@^0.17.0`

### What did you try to do:

I was trying to use 
```tsx
    let options = {pointMatchingDistance: 0.006}
    makerjs.model.combine(modelA, modelB, false, true, false, true, options)
```
instead of just `makerjs.model.combineUnion(modelA, modelB)`.

### What happened:

`TypeError: Cannot read property 'low' of undefined`

### Stack trace:

```
Uncaught TypeError: Cannot read property 'low' of undefined
    at Object.isMeasurementOverlapping (index.js:4877)
    at Object.onPath (index.js:3148)
    at walkRecursive (index.js:2898)
    at Object.walk (index.js:2929)
    at Object.onPath (index.js:3159)
    at walkRecursive (index.js:2898)
    at walkRecursive (index.js:2920)
    at walkRecursive (index.js:2920)
    at Object.walk (index.js:2929)
    at breakAllPathsAtIntersections (index.js:3177)
    at Object.combine (index.js:3314)
    at combineModel (DXFAnalyzer.ts:413)
    at recreateModel (DXFViewer.js:153)
    at DXFViewer.js:169
    at commitHookEffectList (react-dom.development.js:22010)
    at commitPassiveHookEffects (react-dom.development.js:22043)
    at HTMLUnknownElement.callCallback (react-dom.development.js:337)
    at Object.invokeGuardedCallbackDev (react-dom.development.js:386)
    at invokeGuardedCallback (react-dom.development.js:439)
    at flushPassiveEffectsImpl (react-dom.development.js:25379)
    at unstable_runWithPriority (scheduler.development.js:701)
    at runWithPriority$2 (react-dom.development.js:12231)
    at flushPassiveEffects (react-dom.development.js:25348)
    at react-dom.development.js:25227
    at workLoop (scheduler.development.js:645)
    at flushWork (scheduler.development.js:600)
    at MessagePort.performWorkUntilDeadline (scheduler.development.js:197)
```

danmarshall, 2020-02-18:
Are you able to share the JSON for modelA & modelB ?

Bersaelor, 2020-02-19:
So, I was trying to create a minimal example.
But when I reduce the size of my shape too much, the error doesn't happen anymore.

In my example I go through several combination steps, and it 
```
ab = combine(a,b, false, true, false, true, options)
abc = combine(ab,c, false, true, false, true, options)
```
and if I remove the options from either step or just combine `a` and `c` the error doesn't occur anymore. So it seams not to be related to the individual shapes as much as to the size.

I could send you the JSON but I'd rather not publish it publicly since it's IP of a client of ours.

danmarshall, 2020-02-22:
@Bersaelor I understand about the IP. Can you tell me if you are using many Bezier curves?

Bersaelor, 2020-02-22:
I actually sent it to the email in your profile already, could be in your spam folder.
And yes, I believe its mostly arcs.

danmarshall, 2020-02-22:
Hi @Bersaelor it appears that one of your models is not a closed geometry. There were a couple of phantom arcs which were duplicated overlapping other similar arcs in the drawing. These are not always visible when rendered. Try finding a chain, then converting that chain to a new model, then combine with that.

Bersaelor, 2020-02-25:
Ok, but shouldn't it work/fail the same way, regardless of using `options` ?

When I don't use the `options` argument everything works fine, and the models combine exactly in the shape I want, even with the phantom arcs.

danmarshall, 2020-03-02:
Yes it should work/fail the same way. Marking this as a bug. 👍