microsoft/maker.js #456: DXF export - Layer name max length

geolehmann, 2020-02-20:
Hello everybody,

it seems there is a strange 30 character limit to to length of the DXF layer names. 
If I use the following:
```
var polyline= new makerjs.models.ConnectTheDots(false, points);
polyline.layer = "somelonglayernamewhichcontainsmorethanthirtycharacters"
var dxf = makerjs.exporter.toDXF(polyline);
```
the generated .dxf file can be imported without problems into AutoCAD.
However, I want to specifiy the color of the layer, so I add the code:

```
var options = {
        layerOptions: {
            "somelonglayernamewhichcontainsmorethanthirtycharacters": {
                color: 2
            }
};
var dxf = makerjs.exporter.toDXF(polyline,options);
```
Now AutoCAD fails to import the dxf file and tells me there is an invalid table entry on the line with the layer name. if I manually edit the dxf file and change the layer name back to a string with < 30 characters, the file can be imported.

Is this a known issue? Are there any solutions for exporting layer with color definition and names with >30 characters? Thanks for help!

danmarshall, 2020-02-21:
According to [this page](https://knowledge.autodesk.com/support/autocad/learn-explore/caas/CloudHelp/cloudhelp/2015/ENU/AutoCAD-Core/files/GUID-CE211679-F264-4128-BE00-BF4E0E7DEA8E-htm.html), a layer name can be up to 255 characters, but you may not use these characters: `< > / \ “ : ; ? * | = ‘` Also, see #432

geolehmann, 2020-02-21:
Thanks for your reply, however, there arent any of these special characters in the layer name, only [a-z,A-Z].

danmarshall, 2020-02-21:
How about in AutoCAD itself? Are you able to name layers with long names, export, then import again?

ru1973, 2020-02-22:
DXF thru version R14 does not support long layer names, and will switch all lower case to UPPER case

geolehmann, 2020-02-24:
Thanks @ru1973, this explains this strange behaviour. Seems like there is no way around this limitation :-(

closed 2020-02-24 by geolehmann