microsoft/maker.js #458: Why do the segments of a curve change when scaling objects?

Bersaelor, 2020-03-01:
Sorry for opening another issue, but I have more questions.

When I import a curve from DXF (via SVGPath), scale it with makerjs, and then export it again using ` makerjs.exporter.toDXF`, it seems like the number of segments change.

In the following picture, the middle is just import, scale, export:
(Scaled smaller left, scaled exactly 1.0 middle, scaled bigger on the right)

[image]

Is this again because of the storing of the curve pieces as ARCs, hence they will be subdivided in between?

danmarshall, 2020-03-02:
Probably it is a matter of precision. Since arcs do not truly match a curve, there is an error distance tolerance: https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/models/BezierCurve.ts#L186

Perhaps you can also scale your bezierAccuracy when you call [makerjs.importer.fromSVGPathData](https://maker.js.org/docs/importing/#SVG%20path%20data)

The default accuracy calculation is [here](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/models/BezierCurve.ts#L496).

Bersaelor, 2020-03-02:
Wow, changing the accuracy more or less solved it, thanks again for your help!

closed 2020-03-02 by Bersaelor