microsoft/maker.js #461: Model accepted by makerjs.exporter.toSVG(), but not by makerjs.exporter.toDXF()

kristianpedersen, 2020-03-18:
Go to https://maker.js.org/docs/basic-drawing/#Models

Change the last two lines of code to this:

```javascript
var dxf = makerjs.exporter.toDXF(model);
console.log(dxf);
```

I would expect a string, but all I'm getting is `undefined`.

danmarshall, 2020-03-18:
Hello, I got some dxf output:
[image]

kristianpedersen, 2020-03-19:
Ah, the blue console does show the string. Chrome's console however just displays `undefined`:

[image]

kristianpedersen, 2020-03-19:
Here's what's happening on some client work I'm doing now:

When I pass an array of lines, it's accepted by toSVG(), but not toDXF(). Are the array items missing something that's needed to be accepted by toDXF()?

Here you can see where it's failling, and the format of the lines (`objects.flat()`).

[image]