microsoft/maker.js #462: Dogbones on chains

3kmch, 2020-04-17:
I recently jumped back into an older ticket https://github.com/microsoft/maker.js/issues/336.

Some imported SVG paths behave very odd. While trying to figure out why it behaves like this, I took your dognone example from here: https://maker.js.org/docs/working-with-chains/#Chain%20dogbone and rotated the second rect by 15 degrees.

[image]

The left chain is perfect. But not the one on the right. ;-) Any idea, why that happens? 

https://maker.js.org/playground/?script=https://gist.githubusercontent.com/3kmch/c1eb398d8bcf6af2dafefc0c04a5b735/raw/8dfdae2388e8e557db0fb05d091b3f20da66d369/dogbones%2520and%2520chains

3kmch, 2020-04-19:
When I import path data from Inkscape [1], then I get this:
https://maker.js.org/playground/?script=https://gist.githubusercontent.com/3kmch/0cf3211776a57e08f50da84a693a58d7/raw/63aabb8f46c865b858fa646784dcc49bc9989481/svg_path_comparison.js

Which is not the correct interpretation. When I extract the path from https://svg-edit.github.io/svgedit/editor/svg-editor.html [2] then the path looks OK. To switch between the different path comment/uncomment lines 6 and 9
 
```
//SVG from Inkscape
var pathData = "m 267.70703,259.78125 c 0.0881,1.56899 0.14345...

//SVG from SVG-edit
//var pathData = "m267.70703,259.78125c0.0881,1.56899 0.14345...
```
[1] https://gist.github.com/3kmch/4b305c7043ee923b8a390ebffd2661ca#file-testshape_inkscape-svg
[2] https://gist.github.com/3kmch/494e2fa2ac373aec6759b9cdd7ea0b21#file-testshape_svg-edit-svg

Do you recommend a specific toolchain for the import process?

danmarshall, 2020-04-24:
I don't think I'm handling smooth curves with more than 4 points. I see the command:
```
c 0.0881,1.56899 0.14345,3.13964 0.16602,4.71094 -0.004,62.32644 -50.47638,112.87855 -112.80274,112.98047
```
Probably I need to add an extra curve. Until then, can you break these into 2 or more pieces?

3kmch, 2020-05-04:
Thank you for your reply :)

> Until then, can you break these into 2 or more pieces?

I'm not shure wheter I understand you correctly.

Perhaps it's better to work with another example. I replaced the example path with a real one, which I converted from DXF to SVG in Inkscape. This version works quite good (exept for the tilted slots).
https://maker.js.org/playground/?script=https://gist.githubusercontent.com/3kmch/2e89a54606ecf2c76ffec53a631b0978/raw/91ed1b81dc72b2123b3bf522ed6f74d964fdef5c/Dogbones_V2.js

To get a working path I did the following:
1. Upload the SVG to https://svg-edit.github.io/svgedit/editor/svg-editor.html
2. Extract the path d element and paste into maker.js.

If I extract the path element directly from Inkscape, then I get this and that won't work:
`m 255,1920 c 0,-40 -34,-80 -76,-80 -42,0 -76,40 -76,80 0,40 34,80 76,80 42,0 76,-40 76,-80 z m -1226,-890 567,980 c 19,40 57,60 99,60 h 277.2 v -290 h 55.6 v 290 h 277.2 c 42,0 80,-20 99,-60 l 567,-980 c 21,-35 21,-80 0,-115 l -139,-241 -245,142 -28,-48 245,-142 -138,-240 c -20,-36 -58,-57 -99,-57 h -1134 c -41,0 -79,21 -99,57 l -138,240 245,142 -28,48 -245,-142 -139,241 c -21,35 -21,80 0,115 z`

I'm really puzzeled by this behaviour. Do you have an idea what's wrong?