microsoft/maker.js #464: TypeError: makerjs.model.addCaption is not a function (

icodk, 2020-06-03:
Get the following error in an otherwise functioning vuejs and makerjs 
`vue.esm.js?efeb:628 [Vue warn]: Error in mounted hook: "TypeError: makerjs.model.addCaption is not a function"`
Had similar issue in the past   #352
Any idea?

icodk, 2020-06-03:
I can do something like this with no error  but no text is shown
```

              mymodel.caption={
                  text: "This is a caption",
                  anchor: new makerjs.paths.Line([0, 50], [1000, 50])
              };
```

danmarshall, 2020-06-03:
Are you using an older version of the library?

icodk, 2020-06-04:
YES !! :-)
Sorry about that 
I use https://cdn.jsdelivr.net/npm/makerjs@0/target/js/browser.maker.js
Works fine with the new

icodk, 2020-06-04:
But the caption text is not exported to DXF nor to PDF. Is opentype being exported ?

icodk, 2020-06-04:
Sorry about that too
It works great in both PDF and DXF

closed 2020-06-04 by icodk