microsoft/maker.js #466: Adding text to a model

danablend, 2020-06-12:
Hello, I am trying to add text to a model, meaning not just rendering the text to the screen but having it on the model. This is important because I want the text exported when I export as .pdf and .dxf. Currently I cannot figure out how to overcome the asynchronous opentype.load function, which is the bottleneck of this issue. How do I overcome this?

`                        this.models = {
                        
                            grid: grid
                        };

                        // Font loading

                        opentype.load('/fonts/mavenpro/MavenPro-Regular.ttf', function (err, font) {
                            if (err) {
                                // Font failed to load
                            } else {
                                var textModel = new makerjs.models.Text(font, 'Hello', 100);

                                makerjs.model.center(textModel, true, false);

                                this.models.text = textModel; // This does not work, because it's asynchronous.
                            }
                        });`

Essentially, I have a modified version of your Window Arch Grid demo, and I want to add text to the model so I can export the text with the window, and update the text too when the playground values are updated, just like the window figure. Thanks.

danmarshall, 2020-06-12:
Hello, are you doing this in the online Playground or in your own app?

danablend, 2020-06-12:
Hey there, I have set up a coding environment which uses the same libraries as the online Playground with a few changes. I want to do this on a normal apache web-server.

danmarshall, 2020-06-12:
You'll need to render *after* you've set the `textModel`. When I say render, I'm assuming you're calling `makerjs.exporter.toSVG()', then adding the svg to the page.

danablend, 2020-06-12:
The thing is, apart from just having to render the text to the website screen, I need the text added to the model. I have an entire model, like your Window Arch Grid, and I would like to add text to this at a very specific position according to the settings made in my local playground, and I also need it to be included in the .pdf and .dxf export for the model (download buttons). I can't exactly seem to figure out the way to make all of these features included, if I were to just render the svg to the screen.

danmarshall, 2020-06-12:
This is one of the pitfalls of [asynchronous programming](https://developer.mozilla.org/en-US/docs/Learn/JavaScript/Asynchronous/Concepts). 
One thing you might do, is load the font first. Once it is loaded, then start your app.

danablend, 2020-06-12:
I see, I had thought of that solution too, but wanted to hear if there was an easier solution beforehand. I'll see what I can do - thanks.

danablend, 2020-06-13:
Hi, I've made it work. Is there any way to rotate a Text object, or making any work-around to have text rotated? Thanks.

danmarshall, 2020-06-14:
Certainly, use [makerjs.model.rotate](https://maker.js.org/docs/intermediate-drawing/#Rotating).

closed 2021-07-17 by danmarshall