microsoft/maker.js #470: childrenOnChain Problem

Byteschmiede, 2020-08-05:
Hey!

I'm playing around with layout.childrenOnChain and have question how to accomplish following task:

There is a curved path id like to use to position my text on it. Just like childrenOnChain.
The only problem is, that this method is built for text who covers the whole chain.

Example:

Text: Foobaafooba
Text on Path: F_o_o_b_a_a_f_o_o_b_a

Text: Foo
Text on Path: F___________o__________o

But I want to center on the path like:

Text on Path: __________Foo__________

here are Screenshots from the playground

[image]

I hack fix it with pre- and post characters with spaces:
But that's too hacky for my choice

[image]

danmarshall, 2020-08-05:
Hi @Byteschmiede - can you "clip" your chain? Perhaps do a boolean subtraction from a rectangle to get the piece you want.

Byteschmiede, 2020-08-07:
[image]

Hey Dan,

I tried my best but clipping a chain with a rect results to an endless chain.

danmarshall, 2020-08-07:
Hi @Byteschmiede , this is one of those situations where you can use the [flags for boolean operations](https://maker.js.org/docs/advanced-drawing/#Combining%20with%20Boolean%20operations):

```js
var m = require('makerjs');

var curve = new m.models.BezierCurve([[0, 100], [100,0], [200, 100]]);
var square = m.model.move(new m.models.Square(80), [60, 10]);
var rect = new m.models.Rectangle(200, 100);

this.models = { curve, square, rect };

m.model.combine(curve, square, true, false, false, false, {trimDeadEnds: false});
```
[image]
[image]

(the rectangle is just there because of a bug in the Playground canvas 😕)
Also, maybe the documentation shouldn't say _Each model must be a closed geometry_ like it does. In this advanced case you can control which segments are kept and deleted.