microsoft/maker.js #473: how to draw a model with dynamic input

parinya-pusree, 2020-10-16:
Hi from Thailand. 
just getting start with maker.js and new to Github too.
I want to make web app that can create box with finger joint for Laser cut in DXF file. 
I try draw a easy rectangle. 
Output normal when I define width and height and code. like width=300, height=500 
but 
when get values from input box. Output always be in "L" shape. 
I don't know what is going on.
Here is my code : [https://jsfiddle.net/parinya_pusree/z5y70vw4/1/](url)

Any help would be appreciate. Thanks.

danmarshall, 2020-10-20:
Hello, and welcome to Maker.js :)
Your input values are coming in as strings. Maker.js will require these to be numeric. 
I changed your code by coercing to numeric with the `+` operator:
```js
var w = +document.getElementById("boxwidth").value;
var h = +document.getElementById("boxheight").value;
```
Of course, this only works when the values within the textbox are actually numbers. Here is a great article showing several other ways of converting to number and their outcomes:
https://medium.com/better-programming/what-is-the-best-way-to-convert-a-string-to-a-number-in-javascript-67052a1706c6

parinya-pusree, 2020-10-23:
> Hello, and welcome to Maker.js :)
> Your input values are coming in as strings. Maker.js will require these to be numeric.
> I changed your code by coercing to numeric with the `+` operator:
>
> ```js
> var [...]

It is work. Thank you very much.

closed 2020-10-23 by parinya-pusree