microsoft/maker.js #474: Draw a dashed circle using Maker.js.

Salot13, 2020-11-03:
I need to draw a dashed circle using Maker.js. I see all documentation for that in Maker.js but there's no any model for draw it. Is there's any way to draw it?

please see this image for dashed circle which i wants to draw.

[image]

danmarshall, 2020-11-04:
Hello, you can draw a series of arcs:
```
var makerjs = require('makerjs');

function dashedCircle(count, ratio) {

  this.paths = {};
  
  var div = 360 / count;
  var span = div * ratio;
  
  for (var i = 0; i < count; i++) {
    var start = i * div;
    this.paths[i] = new makerjs.paths.Arc([0, 0], 100, start, start + span);
  }
  
}

dashedCircle.metaParameters = [
  { title: "count", type: "range", min: 1, max: 100, value: 40 },
  { title: "ratio", type: "range", min: 0.1, max: 1, value: 0.5, step: 0.1 }
];

module.exports = dashedCircle;
```

Salot13, 2020-11-04:
HI @danmarshall 
Thanks for the help, I have same issue in dashed line. There's any code for that in maker.js?

please see this image for dashed Line which I wants to draw.

[image]

danmarshall, 2020-11-07:
Hello, a dashed line can be drawn like this:
```js
var makerjs = require('makerjs');

function dashedLine(count, ratio) {

  this.paths = {};
  
  //use any line as example
  var line = new makerjs.paths.Line([0, 0], [100, 100]);
  
  var length = makerjs.measure.pathLength(line);
  var totalDashLength = length * ratio;
  var totalGapLength = length - totalDashLength;
  var dashLength = totalDashLength / count;
  var gapLength = totalGapLength / (count - 1);
  var normal = makerjs.point.subtract(line.end, line.origin);
  var dash = new makerjs.paths.Line([0, 0], makerjs.point.scale(normal, dashLength / length)); 
  var gap = makerjs.point.scale(normal, gapLength / length);
  
  var origin = line.origin;
  for (var i = 0; i < count; i++) {
    var d = makerjs.$(dash)
      .clone()
      .move(origin)
      .addTo(this, i)
      .$result;
    origin = makerjs.point.add(d.end, gap);
  }
  
}

dashedLine.metaParameters = [
  { title: "count", type: "range", min: 2, max: 100, value: 10 },
  { title: "ratio", type: "range", min: 0.1, max: 1, value: 0.6, step: 0.1 }
];

module.exports = dashedLine;
```

Salot13, 2020-11-19:
HI @danmarshall

In first quadrant I have co-ordinates for start and end points. I want to draw a dashed arc between these points.  
I have the following points -
Start = 608 , End = 669 , Radius = 638.5, origin =[0,0]

According to the documentation, these are the mentioned ways to draw an arc. 
Method  1 => new Arc(pointA: IPoint, pointB: IPoint, radius: number, largeArc: boolean, clockwise: boolean): Arc
Method 2 => new Arc(origin: IPoint, radius: number, startAngle: number, endAngle: number): Arc

According to my use case, which method should I use and how?

danmarshall, 2021-07-17:
To accomplish this with arcs, here is a variation of the circle code from above:
```js
var makerjs = require('makerjs');

function dashedArc(count, ratio, startAngle, endAngle) {

  this.paths = {};
  
  var sweep = endAngle - startAngle;
  var div = sweep / count;
  var span = div * ratio;
  
  for (var i = 0; i < count; i++) {
    var start = startAngle + i * div;
    this.paths[i] = new makerjs.paths.Arc([0, 0], 100, start, start + span);
  }
  
}

dashedArc.metaParameters = [
  { title: "count", type: "range", min: 1, max: 100, value: 40 },
  { title: "ratio", type: "range", min: 0.1, max: 1, value: 0.5, step: 0.1 },
  { title: "startAngle", type: "range", min: 0, max: 360, value: 45 },
  { title: "endAngle", type: "range", min: 0, max: 360, value: 200 },
];

module.exports = dashedArc;
```

closed 2021-07-17 by danmarshall