microsoft/maker.js #475: PDF Lineweight

PrimalOrB, 2020-11-12:
Is there any ability to define line weight / stroke width for a maker.js object through the pdfkit export. 
The only option I can see is to define the stroke color within the IPDFRenderOptions

danmarshall, 2020-11-20:
Hello, sorry it took me so long to get back to you. I have been looking into this. I think we should be able to add `layerOptions`, the same as the [svg](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/svg.ts#L908) and [dxf](https://github.com/microsoft/maker.js/blob/master/packages/maker.js/src/core/dxf.ts#L539) exporters.

PrimalOrB, 2020-11-20:
not a problem, thanks for looking into it. It seems like it would be a good addition

-regards

kiranm211, 2026-10-13:
Hello, this is @kiranm211. 
I would like to take on this issue, as I currently need support for line weight in my project.

danmarshall, 2026-11-09:
@kiranm211 if you can send a PR I will take a look at it!