microsoft/maker.js #479: Can marker.js convert svg to DXF？

salamander-mh, 2020-12-28:
any simple demo？

tabeaeggler, 2021-01-04:
Here you can find my approach which is not fully working yet: https://github.com/microsoft/maker.js/issues/480

danmarshall, 2021-01-04:
Hello, first it is important to note that Maker.js only deals with "outlines" - drawings made of lines, circles, arcs, and Bezier Curves. SVG and DXF both have many concepts beyond this, such as color, fill, line thickness, etc. So during import you would lose some of this information. Also, not every concept in SVG has a counterpart in DXF, so might also lose some features in the translation.
Does your SVG contain any symbols, or use any repeating patterns as a fill? This will be difficult to reconstruct...
But if you are strictly importing / exporting lines, circles and arcs, it may work.

tabeaeggler, 2021-01-06:
@salamander-mh I posted a working version https://github.com/microsoft/maker.js/issues/480

z3dev, 2021-01-06:
@salamander-mh you might try some other libraries, or applications. There are lots of applications converting between SVG and DXF.

salamander-mh, 2021-01-11:
@tabeaeggler  how to deal with `path` element with BezierCurve?

salamander-mh, 2021-01-11:
@z3dev  Can you recommend some applications?  thks a lot.

tabeaeggler, 2021-01-11:
> @tabeaeggler how to deal with `path` element with BezierCurve?

1. Get the d attribute: let d = path.getAttribute("d")
2. Create a makerjs model: let svgPathData = makerjs.importer.fromSVGPathData(d). It will automatically create a model with different curves representing a BezierCurve. It is structured as follows:  "models", "p_1", "models", "Curve_1", "Curve_2", "Curve_3", "Curve_4".
3. Add it to your model: makerjs.model.addModel(model, svgPathData, "type"). Print your svgPathData and figure out your type.  For my semi-ellipses I used the type "ellipticarc"

closed 2021-07-17 by danmarshall