microsoft/maker.js #487: Problem with SVG code in https://maker.js.org/playground/?script=Belt

ru1973, 2021-02-13:
When I use the default values for the Belt script, and export SVG, the overall size of the shape is not correct. If I export the same values to a DXF file, the size is correct. I am using mm for both.

Below is the first line of SVG code generated. From my testing, it is being exported in pixels. There should be the text mm after the 95 and the 60 - then it exports correct.

<svg width="95" height="60" viewBox="0 0 95 60" xmlns="http://www.w3.org/2000/svg"><g id="svgGroup" stroke-

danmarshall, 2021-02-13:
Hello @ru1973 , units are not assumed. So everything is unitless by default in Maker.js. When you export to any specific format, that format's default unit system is assumed. In SVG, I believe it is pixels. In DXF I believe it is inches (not exactly sure).
But if you set your drawing's units then it will export as that unit system. 
See https://maker.js.org/docs/basic-drawing/#Units and https://maker.js.org/docs/exporting/#content

ru1973, 2021-02-13:
Thanks Dan. I will add the mm to the SVG export.

ru1973, 2021-02-13:
Hi Dan,

I think this is where I got confused. The screen says that the exported data is in millimeters.
[image]

ru1973, 2021-02-15:
Hi Dan,

I only reopened it to show you what is being displayed in playground for export. Is there an easy way to add mm to code in SVG file when you do a copy and paste in playground? I can do it in a text editor before I paste it into the program I use for laser.

<svg width="95" height="60" 

to be <svg width="95mm" height="60mm" 

I have been using the technique to copy the SVG code from the playground and pasting it into a laser program called Lightburn. I then scale the file in Lightburn from pixels to mm. I only started doing it because it was a bit quicker than saving a DXF and importing it into Lightburn. Maker.js exports unitless to DXF, but because Maker.js is using mm, it sizes correctly in Lightburn.  If it isn't easily doable, I will ask the developer of Lightburn if he can add the code to his SVG import when the units aren't identified with mm, Inch, etc.

Ralph

danmarshall, 2021-02-16:
Hi Ralph,
Thanks for pointing this out. In the playground my intention was to have a unit chooser at the time of export. As of now, it only has the Accuracy setting:

[image]

Is this the part you are referring to - having an additional Units options here?

ru1973, 2021-02-16:
Yes. A units option would be nice. I think it would add flexibility to the export option. Most people that are "makers" either use mm or imperial units. But, I am sure that there are other users who want to create SVG icons etc that would like the shapes to be in pixels.