microsoft/maker.js #515: grid checkbox reversed

hrgdavor, 2022-01-11:
current meker.js website

[image]

[image]

danmarshall, 2022-01-11:
I've seen this also, sometimes. Not sure, but it seems the browser is trying to be smart about caching the value??

hrgdavor, 2022-01-11:
You are right, it could be that if code is not checking checkbox value but depending on internal state,  and browsers to tend to try to restore input value sometimes on refresh. ... it is probably something weird like that...

hrgdavor, 2022-01-12:
ok, found a way to reproduce. Tested in Chrome and Brave (just in case maybe is there difference) and firefox
steps added to main text aboeve