microsoft/maker.js #520: Duplicate paths on single line font in DXF export

timdiacon, 2022-02-08:
I'm using a single line font to create text objects which are then engraved on a CNC machine. I'm getting an issue where the paths of the font are duplicated in the exported DXF file.

I have two versions of the font. The first uses beziers as below

[image]

[image]

The second image above is a screen shot from a CAD application where I have offset the duplicate lines. You will see each individual bezier occurs twice in the file. I have also selected every other path in the duplicates which then appear red and you can see there are a lot more beziers than in the original font (i have tried experimenting with the bezier accuracy option but it doesn't seem to effect the output), this is a minor point but lead me to try using a version of the font which is just comprised of straight lines as below

[image]

Again each individual line is duplicated in the export as shown below however this time the straight lines match those of the font

[image]

The duplication is a big issue as when the DXF is imported into out toolpathing application it treats the duplicate lines as an object which needs to be closed and preventing us from then joining them together into a single line.

I think this duplication is a bug caused by single line fonts not being closed fills like more regular fonts.

As a side note I've also tried using captions instead but we get problems where these are not parsed correctly in the toolpathing application and get strangely rotated. We can't also measure the size and then position captions as accurately as we can text objects so we really want to try and fix this duplication. If we manually open up the DXF and remove the duplicates and resave it is imported perfectly into the tool pathing application as a series of open vectors which can then be joined together.

timdiacon, 2022-02-08:
p.s. happy to offer a bounty to get this fixed as would save us a huge amount of machining time

z3dev, 2022-02-08:
So… no layers… no grouping… just a single letter from a single line font?

Can you attach the font?

danmarshall, 2022-05-12:
Hello @timdiacon - my apologies that I just got to this, I just found it in my inbox. I loaded the ttf file you sent, and it does indeed contain a double line on top of itself. I surmise that this was the font developer's method of creating a "closed" geometry with zero area.

If you cannot get a font that is truly a single line, you may need to write a function to iterate over paths of a Text model and remove duplicates by comparison.